YosysHQ/yosys #2596: assertpmux pass generates wrong assertions for priority case

msinger, 2021-02-22:
## Steps to reproduce the issue
test.sv:
```
module test(input logic clk, output logic out1, out2, input logic [1:0] foo);

	assume property (foo == 3);

	always_ff @(posedge clk) begin
		priority case (1)
			foo == 3: out1 = 0;
			foo[0]:   out1 = 1;
			default;
			// should NOT assert, because first case (foo==3) matched and also because default case exists
		endcase

		priority case (foo)
			2: out2 = 0;
			1: out2 = 1;
			// should assert, because case foo==3 doesn't exist
		endcase
	end

endmodule
```

test.sby:
```
[tasks]
test

[options]
mode bmc
append 10
expect pass
multiclock on
depth 20

[engines]
smtbmc

[script]
read_verilog -sv -formal test.sv
prep -top test
assertpmux

[files]
test.sv
```
Or download:
[test.zip](https://github.com/YosysHQ/yosys/files/6018876/test.zip)

Run:
`sby test.sby`

## Expected behavior

The assertion for the first priority case should not trigger, because foo is 3 and that case exists. Also there is a default case, which in my understanding should never cause an assertion to trigger.

The assertion for the second priority case should trigger, because there is no case for foo==3 and the default case doesn't exist.

## Actual behavior

The opposite happens: The assertion for the first priority case triggers and the one for the second does not.

mwkmwkmwk, 2021-02-22:
Ugh.  So there are two issues here.

First is that, somehow (okay I know how and it is horrible), yosys actually parses `priority` as `unique`, and this is an actual bug that needs fixing.

Second is that `assertpmux` is not actually doing what you think it's doing — it basically verifies the uniqueness of the triggered case (ie. checks validity of `parallel_case`), but it doesn't check if the default case is triggered in any way (so it should do nothing for `priority` cases, but do some incomplete checking for `unique` cases).

msinger, 2021-02-22:
Oh, that means checking for `full_case` would be a new feature. Only the first case is actually a bug, where it checks the `priority case` for uniqueness.

commit 12e699f0 (2021-02-22): Fix handling of unique/unique0/priority cases in the frontend.

Basically:

- priority converts to (* full_case *)
- unique0 converts to (* parallel_case *)
- unique converts to (* parallel_case, full_case *)

Fixes #2596.
PR #2597 (2021-02-25): Fix handling of unique/unique0/priority cases in the frontend.
Basically:

- priority converts to (* full_case *)
- unique0 converts to (* parallel_case *)
- unique converts to (* parallel_case, full_case *)

Fixes #2596.
commit a651204e (2021-02-25): Fix handling of unique/unique0/priority cases in the frontend.

Basically:

- priority converts to (* full_case *)
- unique0 converts to (* parallel_case *)
- unique converts to (* parallel_case, full_case *)

Fixes #2596.
closed 2021-02-25 by mwkmwkmwk