YosysHQ/yosys #2600: No longer able to use write_smt2 with circuits that have async-reset FFs (without clk2fflogic)

nmoroze, 2021-02-23:
I have a project that calls the `write_smt2` backend on a circuit that has asynchronously reset flip-flops. I was originally using techmap to replace them with an equivalent circuit, but that no longer works after commit 09071afe. Here's an example input file:
```verilog
module test(
    input clk,
    input rst,
    input d,
    output reg q
);
always @(posedge clk or negedge rst) begin
    if (!rst) begin
        q <= 1'b0;
    end else begin
        q <= d;
    end
end
endmodule
```

Running the following:
```
read_verilog test.v
prep
techmap -map +/adff2dff.v
```
Gives:
```
3.1. Executing Verilog-2005 frontend: /usr/local/bin/../share/yosys/adff2dff.v
Parsing Verilog input from `/usr/local/bin/../share/yosys/adff2dff.v' to AST representation.
/usr/local/bin/../share/yosys/adff2dff.v:14: ERROR: syntax error, unexpected TOK_REG
```

I figured that this approach may now be deprecated in favor of `async2sync`, however this results in a different issue since as of 4a05cad7 async2sync replaces `adff`s with `sdff`s, which are also not supported by the write_smt2 backend. For example, running this on the same input file:
```
read_verilog test.v
prep
async2sync
write_smt2
```

Gives:
```
4. Executing SMT2 backend.
; SMT-LIBv2 description generated by Yosys 0.9+3911 (git sha1 cc7d18d2, clang 12.0.0 -fPIC -Os)
Creating SMT-LIBv2 representation of module test.
ERROR: Unsupported cell type $sdff for cell test.$procdff$3.
```

One way I can get this circuit into a state accepted by write_smt2 is by using clk2fflogic, but this feels a bit heavy-handed to me. 

Is this considered an issue? I'd be happy to help by contributing a PR, but I'm not sure what the best approach to resolving this would be. I could imagine either fixing write_smt2 to transform `sdff` cells, or modifying `async2sync` to no longer emit them (like it was prior to 4a05cad7).

zachjs, 2021-02-23:
Regarding the techmap issue:

The declaration in `adff2dff.v` is illegal. Per Section 6.7.1 of the LRM:
> A net type keyword shall not be followed directly by the reg keyword

I believe we should:
1. Declare `NEXT_Q` as simply a `reg` in `adff2dff.v`
2. Add proper test coverage for the `adff2dff` functionality, if it is still relevant
3. Consider adding coverage that ensures everything in `techlibs` gets past the frontend. It may prudent to run certain files through with, say, multiple `LUT_WIDTH`s

whitequark, 2021-02-23:
> 3\. Consider adding coverage that ensures everything in `techlibs` gets past the frontend. It may prudent to run certain files through with, say, multiple `LUT_WIDTH`s

Agreed, this would be very nice to have!

mwkmwkmwk, 2021-02-23:
Run `dffunmap` after `async2sync` — this will break up the `$sdff`.

mwkmwkmwk, 2021-02-23:
I've submitted #2601, which will at least give a hint on how to solve the problem.

nmoroze, 2021-02-23:
Thanks all for the quick replies and fixes! In particular, thanks @mwkmwkmwk for the pointer to `dffunmap` and including that tip in error messages - that seems super helpful.

PR #2602 (2021-02-24): Fix syntax error in adff2dff.v
Fixes #2600.
commit cde73428 (2021-02-24): Fix syntax error in adff2dff.v

Fixes #2600.
closed 2021-02-24 by mwkmwkmwk