YosysHQ/yosys #2606: yosys crashes on error output when "logger -expect" is used

msinger, 2021-02-24:
I ran `make test` and the tests that use `logger -expect ...` fail, but I guess this only happens for me. Here an example:
```
 Yosys 0.9+3911 (git sha1 f4f471f34, clang 3.4.2 -fPIC -Os)

yosys> logger -expect error oink 1
Added regex 'oink' for warnings to expected error list.

yosys> read_verilog <<EOT
module inverter(input a, output y);
assign y = (a == 1'b0? 1'b1 : 1'b0);
endmodule // inverter
EOT

1. Executing Verilog-2005 frontend: <stdin>
Parsing Verilog input from `<stdin>' to AST representation.
Warning: Yosys has only limited support for tri-state logic at the moment. (<stdin>:2)
<stdin>:2: ERROR: syntax error, unexpected TOK_CONSTVAL
*** Error in `/home/michael/projects/yosys/yosys': double free or corruption (!prev): 0x0000000001799c70 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81329)[0x7f3f535d6329]
/home/michael/projects/yosys/yosys[0x6233b9]
/home/michael/projects/yosys/yosys[0x62329e]
/home/michael/projects/yosys/yosys(_ZN5Yosys13log_cmd_errorEPKcz+0x0)[0x623822]
/home/michael/projects/yosys/yosys(_ZN5Yosys11cover_extraESsSsb+0x0)[0x624775]
/home/michael/projects/yosys/yosys[0x6233c8]
/home/michael/projects/yosys/yosys(_ZN5Yosys14log_file_errorERKSsiPKcz+0xb7)[0x6234cf]
/home/michael/projects/yosys/yosys[0x727625]
/home/michael/projects/yosys/yosys(_Z24frontend_verilog_yyparsev+0x7a0)[0x70a8fc]
/home/michael/projects/yosys/yosys(_ZN5Yosys15VerilogFrontend7executeERPSiSsSt6vectorISsSaISsEEPNS_5RTLIL6DesignE+0x11d9)[0x729cbb]
/home/michael/projects/yosys/yosys(_ZN5Yosys8Frontend7executeESt6vectorISsSaISsEEPNS_5RTLIL6DesignE+0xb2)[0x568dac]
/home/michael/projects/yosys/yosys(_ZN5Yosys4Pass4callEPNS_5RTLIL6DesignESt6vectorISsSaISsEE+0x1b6)[0x5680fe]
/home/michael/projects/yosys/yosys(_ZN5Yosys4Pass4callEPNS_5RTLIL6DesignESs+0x51a)[0x567d80]
/home/michael/projects/yosys/yosys(_ZN5Yosys5shellEPNS_5RTLIL6DesignE+0x170)[0x62e327]
/home/michael/projects/yosys/yosys(main+0xb1a)[0x561055]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f3f53577555]
/home/michael/projects/yosys/yosys[0x56040d]
```

If I run it with valgrind, this is what I get after entering the last line:
```
1. Executing Verilog-2005 frontend: <stdin>
Parsing Verilog input from `<stdin>' to AST representation.
Warning: Yosys has only limited support for tri-state logic at the moment. (<stdin>:1)
<stdin>:1: ERROR: syntax error, unexpected TOK_CONSTVAL
==23247== Invalid free() / delete / delete[] / realloc()
==23247==    at 0x4C2B06D: free (vg_replace_malloc.c:540)
==23247==    by 0x6233B8: Yosys::logv_error_with_prefix(char const*, char const*, __va_list_tag*) (log.cc:349)
==23247==    by 0x62329D: Yosys::logv_error(char const*, __va_list_tag*) (log.cc:372)
==23247==    by 0x623821: Yosys::log_error(char const*, ...) (log.cc:434)
==23247==    by 0x624774: Yosys::log_check_expected() (log.cc:710)
==23247==    by 0x6233C7: Yosys::logv_error_with_prefix(char const*, char const*, __va_list_tag*) (log.cc:356)
==23247==    by 0x6234CE: Yosys::log_file_error(std::string const&, int, char const*, ...) (log.cc:382)
==23247==    by 0x727624: frontend_verilog_yyerror(char const*, ...) (verilog_frontend.cc:651)
==23247==    by 0x70A8FB: frontend_verilog_yyparse() (verilog_parser.tab.cc:7457)
==23247==    by 0x729CBA: Yosys::VerilogFrontend::execute(std::istream*&, std::string, std::vector<std::string, std::allocator<std::string> >, Yosys::RTLIL::Design*) (verilog_frontend.cc:480)
==23247==    by 0x568DAB: Yosys::Frontend::execute(std::vector<std::string, std::allocator<std::string> >, Yosys::RTLIL::Design*) (register.cc:466)
==23247==    by 0x5680FD: Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::string, std::allocator<std::string> >) (register.cc:316)
==23247==  Address 0x6d34840 is 0 bytes inside a block of size 400 free'd
==23247==    at 0x4C2B06D: free (vg_replace_malloc.c:540)
==23247==    by 0x6233B8: Yosys::logv_error_with_prefix(char const*, char const*, __va_list_tag*) (log.cc:349)
==23247==    by 0x6234CE: Yosys::log_file_error(std::string const&, int, char const*, ...) (log.cc:382)
==23247==    by 0x727624: frontend_verilog_yyerror(char const*, ...) (verilog_frontend.cc:651)
==23247==    by 0x70A8FB: frontend_verilog_yyparse() (verilog_parser.tab.cc:7457)
==23247==    by 0x729CBA: Yosys::VerilogFrontend::execute(std::istream*&, std::string, std::vector<std::string, std::allocator<std::string> >, Yosys::RTLIL::Design*) (verilog_frontend.cc:480)
==23247==    by 0x568DAB: Yosys::Frontend::execute(std::vector<std::string, std::allocator<std::string> >, Yosys::RTLIL::Design*) (register.cc:466)
==23247==    by 0x5680FD: Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::string, std::allocator<std::string> >) (register.cc:316)
==23247==    by 0x567D7F: Yosys::Pass::call(Yosys::RTLIL::Design*, std::string) (register.cc:293)
==23247==    by 0x62E326: Yosys::shell(Yosys::RTLIL::Design*) (yosys.cc:1252)
==23247==    by 0x561054: main (driver.cc:512)
==23247==  Block was alloc'd at
==23247==    at 0x4C29F73: malloc (vg_replace_malloc.c:309)
==23247==    by 0x5679B28: xmalloc (xmalloc.c:61)
==23247==    by 0x567A08B: add_history (history.c:293)
==23247==    by 0x62E288: Yosys::shell(Yosys::RTLIL::Design*) (yosys.cc:1239)
==23247==    by 0x561054: main (driver.cc:512)
==23247== 
==23247== 
==23247== HEAP SUMMARY:
==23247==     in use at exit: 824,316 bytes in 3,236 blocks
==23247==   total heap usage: 7,364 allocs, 4,129 frees, 1,117,501 bytes allocated
==23247== 
==23247== 28 bytes in 1 blocks are definitely lost in loss record 395 of 1,928
==23247==    at 0x4C29F73: malloc (vg_replace_malloc.c:309)
==23247==    by 0x5679B28: xmalloc (xmalloc.c:61)
==23247==    by 0x56602FC: readline_internal_teardown (readline.c:501)
==23247==    by 0x56610E1: readline_internal (readline.c:652)
==23247==    by 0x56610E1: readline (readline.c:395)
==23247==    by 0x62E223: Yosys::shell(Yosys::RTLIL::Design*) (yosys.cc:1225)
==23247==    by 0x561054: main (driver.cc:512)
==23247== 
==23247== LEAK SUMMARY:
==23247==    definitely lost: 28 bytes in 1 blocks
==23247==    indirectly lost: 0 bytes in 0 blocks
==23247==      possibly lost: 0 bytes in 0 blocks
==23247==    still reachable: 824,288 bytes in 3,235 blocks
==23247==                       of which reachable via heuristic:
==23247==                         stdstring          : 184,275 bytes in 1,413 blocks
==23247==         suppressed: 0 bytes in 0 blocks
==23247== Reachable blocks (those to which a pointer was found) are not shown.
==23247== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==23247== 
==23247== For lists of detected and suppressed errors, rerun with: -s
==23247== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
```

zachjs, 2021-02-24:
On my machine, the tests that use `logger -expect error` do pass. Which tests are failing for you? There does appear to be an issue when there is an _unmatched_ expected error pattern (as in the above example `oink` of course doesn't match the given parse error). I will fix this double-free issue shortly, but I think there may be another issue at hand causing the patterns not to match for you.

msinger, 2021-02-24:
I ran ./run-tests.sh in each subdir of ./tests that contains this shell script. This is where it failed:
- opt/opt_clean_init.ys (`logger -expect warning` didn't match, so it's not only errors)
- select/unset2.ys
- smv/run_single.sh (NuSMV: command not found)
- svtypes/multirange_subarray_access.ys
- various/bug1745.ys
- verilog/block_labels.ys

I removed the line that frees the readline history, which caused the double free that you fixed properly. Just for testing of course. Now it doesn't crash anymore, but you are right, for some reason the matching of the messages doesn't work for me. I'll try to debug a bit. Maybe it's some incompatibility with my system libs. I use CentOS 7.

msinger, 2021-02-24:
Ok, I kind of have a picture. CentOS 7 comes with GCC 4.8.5. I have also installed Clang 3.4.2 from the CentOS extras repo. I built Yosys with Clang.

In Yosys kernel/log.h there is a comment about GCC 4.8:
```
// In GCC 4.8 std::regex is not working correctlty, in order to make features
// using regular expressions to work replacement regex library is used
#if defined(__GNUC__) && !defined( __clang__) && ( __GNUC__ == 4 && __GNUC_MINOR__ <= 8)
    #include <boost/xpressive/xpressive.hpp>
...
```
I wrote a small test:
```
#include <iostream>
#include <string>
#include <regex>
int main(int argc, char **argv)
{
	std::regex re("syntax error, unexpected TOK_CONSTVAL");
	std::string err = "syntax error, unexpected TOK_CONSTVAL\n";
	std::cout << std::regex_search(err, re);
	return 0;
}
```
When I build this with Clang 3.4.2 the regex does not match, just like my problem in Yosys. I found out that Clang 3.4.2 from the official CentOS repository does not have its own regex standard header, it uses the one from GCC 4.8.5. Surprisingly, this small test works when I compile it with GCC 4.8.5, the regex matches.

Since there is already a check for GCC 4.8 in kernel/log.h, I don't know what else could be done on Yosys side. I already had installed GCC 7.3.1, because Nextpnr requires something newer. I will build Yosys with this one too from now on instead of Clang. That works for me.

zachjs, 2021-03-17:
Perhaps the clang exclusion in `kernel/log.h` should be removed? Whether the regex implementation is broken should depend on the glibc version, not the compiler version. I'm not sure why that was added in #1766. Thoughts, @mmicko?

msinger, 2021-03-17:
@zachjs, I think it's a bit more complicated. If I understand it correctly, then std::regex is part of libstdc++, not glibc. And libstdc++ is part of the GCC sources, but Clang also provides its own implementation for libstdc++. Since you can have only one of them installed in the standard directory /usr/lib, the RHEL team seems to have decided to only ship the GCC version of that lib, not the one from Clang. This means ``kernel/log.h`` would have to check from which "vendor" the libstdc++ library comes from, and if it's GCC, then check if it is newer than 4.8. The problem is, that checking if the compiler is GCC does not help, like in my case. I used Clang, but on CentOS7/RHEL7 only the GCC version of libstdc++ is available. So even when the compiler is Clang, it still links to the faulty libstdc++ from GCC 4.8 (or uses the templates from their headers in /usr/include/c++/4.8.\*/bits/regex\*).

I think the correct thing to do is to check the version of libstdc++. Seems like this is a bit of a mess though. I found something here:
https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html#abi.versioning
If the GCC version of the lib is used, then ``__GLIBCXX__`` should be defined. Some older versions though use the name ``__GLIBCPP__`` instead. So
```
#if defined(__GLIBCPP__) || defined(__GLIBCXX__)
```
should be true if the GCC version of libstdc++ is used. The macro contains the release date of libstdc++, not the version number. So the newest release of GCC 4.8, which is 4.8.5, has ``__GLIBCXX__`` defined to ``20150623``. GCC 4.9.0 came out in 2014. This means that checking for ``__GLIBCXX__ <= 20150623`` would also match for most versions of GCC 4.9.x and even early versions of GCC 5, which according to the comment in ``kernel/log.h`` don't have the regex problem. See here:
https://gcc.gnu.org/develop.html#timeline

To be sure, I tried this:
```
#include <iostream>
#include <regex>

using namespace std;

int main(int argc, char** argv)
{
#ifdef _GLIBCXX_RELEASE
	cout << "_GLIBCXX_RELEASE: " << _GLIBCXX_RELEASE << endl;
#endif
#ifdef _GLIBCPP_VERSION
	cout << "_GLIBCPP_VERSION: " << _GLIBCPP_VERSION << endl;
#endif
#ifdef __GLIBCXX__
	cout << "__GLIBCXX__: " << __GLIBCXX__ << endl;
#endif
#ifdef __GLIBCPP__
	cout << "__GLIBCPP__: " << __GLIBCPP__ << endl;
#endif
	return 0;
}
```
I compiled it with both compilers:
```
g++ -std=c++11 -o test_gcc test.cc
clang++ -std=c++11 -o test_clang test.cc
```
Both, ``test_gcc`` and ``test_clang``, give the same output:
```
__GLIBCXX__: 20150623
```
(The other macros I've tested, ``_GLIBCXX_RELEASE`` and ``_GLIBCPP_VERSION`` and ``__GLIBCPP__`` are not defined.)

mmicko, 2021-03-17:
@msinger This is good approach. I would need to check multiple compiler versions to confirm it works fine in expected cases. If you are willing please prepare PR for this. Thanks

msinger, 2021-03-17:
@mmicko Ok, I'll prepare a PR. I also checked to make sure that libstdc++ doesn't just call some regex function from glibc. I compiled the regex test from above with three different compilers on my system:
```
#include <iostream>
#include <string>
#include <regex>
int main(int argc, char **argv)
{
	std::regex re("syntax error, unexpected TOK_CONSTVAL");
	std::string err = "syntax error, unexpected TOK_CONSTVAL\n";
	std::cout << std::regex_search(err, re);
	return 0;
}
```
```
clang++ -std=c++11 -o test_clang test.cc
g++ -std=c++11 -o test_gcc48 test.cc
scl enable devtoolset-7 bash    # <---- setup GCC 7.3.1
g++ -std=c++11 -o test_gcc7 test.cc
```
``test_clang`` and ``test_gcc48`` both fail the test. They print ``0`` instead of the correct result ``1``. ``test_gcc7`` prints ``1``. If I check them with ``ldd``, all three binaries link the same set of libraries (same glibc and same libstdc++). I think this means that the bug does not reside in the .so files. It must be the header files, because GCC 4.8.5 and Clang both include the regex header from GCC 4.8.5. GCC 7.3.1 includes its own version of that header. Those headers contain a lot of cryptic template stuff I don't understand. I think the GCC 4.8.5 headers are the problem and therefore checking the ``_GLIBCXX*`` type of macros defined in those headers should work.
I'm not totally sure if there isn't any mistake in my reasoning, because I don't understand in detail how the regex stuff is working in those templates. I find it weird that they would implement this again, even though glibc already contains some regex functions for C, if I remember correctly. Please, if somebody knows more about this, point out if I made a mistake somewhere that could compromise my conclusion.

PR #2681 (2021-03-19): Fix check for bad std::regex
Fixes #2606

I tested it with GCC 4.8.5, GCC 7.3.1 and Clang 3.4.2 on CentOS 7. All three produce a binary with a working ``logger -expect`` command. I tested each with these yosys commands:
```
logger -expect error "syntax error, unexpected TOK_CONSTVAL" 1
read_verilog <<EOT
module inverter(input a, output y);
assign y = (a == 1'b0? 1'b1 : 1'b0);
endmodule // inverter
EOT
```
Yosys terminates with exit code 0. And then again with a non-matching string for the ``logger`` command: Yosys terminates with exit code 1.
commit d05d47cc (2021-03-19): Fix check for bad std::regex (fixes #2606)
closed 2021-03-19 by mmicko