YosysHQ/yosys #2608: clock generation error during cover formal analysis

promach, 2021-02-24:
## Steps to reproduce the issue

[clock_gating_post_control.zip](https://github.com/YosysHQ/yosys/files/6035885/clock_gating_post_control.zip)

```
sby -f clock_gating_post_control.sby
gtkwave clock_gating_post_control.gtkw
```
```
`ifdef FORMAL
reg first_clock_had_passed;
initial first_clock_had_passed = 0;
always @(posedge clk) first_clock_had_passed <= 1;

initial assume(clk == 0);

always @($global_clock)
begin
	if(first_clock_had_passed)
		assume(clk == ~$past(clk));
end

always @(posedge clk)
	assume(enable == 0);

localparam MAX_LIMIT = 16;
reg [$clog2(MAX_LIMIT)-1:0] count;
initial count = 0;

always @(posedge clk)
begin
	count <= count + 1;
end

localparam STARTS_FALLING = 4;
localparam STARTS_RISING = 8;

always @(posedge clk)
begin
	// scan_enable signal is asserted low for (STARTS_RISING-STARTS_FALLING-1) clock cycles
	if((count > STARTS_FALLING) && (count < STARTS_RISING))  
		assume(scan_enable == 0);
		
	else
		assume(scan_enable == 1);
end

always @(posedge clk)
begin
	if(first_clock_had_passed)
		cover(~gclk && $past(gclk));
end
`endif

```

## Expected behavior

`count` signal should **ONLY** increment during `posedge clk` , not during `negedge clk`

## Actual behavior

[image]

mwkmwkmwk, 2021-02-24:
By default, yosys formal verification operates in 'single clock domain' mode where clocks in all FFs are ignored, instead using the global clock — thus removing needless detail of simulating the clock signal from the verification process.  If you have multiple clocks, please use `clk2fflogic` to enable actual multi-clock formal verification.

closed 2021-02-24 by mwkmwkmwk

promach, 2021-02-24:
@mwkmwkmwk  how is `clk2fflogic` different from `multiclock on` option in sby ?

mwkmwkmwk, 2021-02-24:
It's the same thing — `multiclock on` in sby enables the `clk2fflogic` pass in yosys.

promach, 2021-02-24:
@mwkmwkmwk Just a beginner question, why do I need `multiclock on` option when the code is **ONLY** using the generated `clk` from `$global_clock` ?

promach, 2021-02-24:
I replaced `always @(posedge clk)` with `always @($global_clock)` , and using `multiclock on` option, but the `count` issue still persists. Why ?

```
`ifdef FORMAL
reg first_clock_had_passed;
initial first_clock_had_passed = 0;
always @($global_clock) first_clock_had_passed <= 1;

always @($global_clock)
	assume(enable == 0);

localparam MAX_LIMIT = 16;
reg [$clog2(MAX_LIMIT)-1:0] count;
initial count = 0;

always @($global_clock)
begin
	count <= count + 1;
end

localparam STARTS_FALLING = 4;
localparam STARTS_RISING = 8;

always @($global_clock)
begin
	// scan_enable signal is asserted low for (STARTS_RISING-STARTS_FALLING-1) clock cycles
	if((count > STARTS_FALLING) && (count < STARTS_RISING))  
		assume(scan_enable == 0);
		
	else
		assume(scan_enable == 1);
end

always @($global_clock)
begin
	if(first_clock_had_passed)
		cover(~gclk && $past(gclk));
end
`endif
```

promach, 2021-02-24:
[Problem solved](https://github.com/YosysHQ/yosys/files/6036102/clock_gating_post_control.zip)  by using `$stable()` on the input signals.

[image]

promach, 2021-03-12:
@mwkmwkmwk On line #129 of the verilog code, how do I `assume()` that the `scan-enable` signal to be **_asynchronous_** to the FPGA/chip system clock ?

I tried the following 2 methods, but they failed the `cover()` check ?

```
// scan-related test signal will never be synchronous to the chip clock
always @($global_clock) if (!$fell(clk)) assume($stable(scan_enable));
```

```
// scan-related test signal will never be synchronous to the chip clock
always @($global_clock) if (!$rose(clk)) assume(!$stable(scan_enable));
```