YosysHQ/yosys #2612: Undescriptive error message when using signal name as a label name

jfierro, 2021-02-25:
[example.txt](https://github.com/YosysHQ/yosys/files/6045737/example.txt)

## Steps to reproduce the issue

Rename example.txt as example.sby then run `sby -f example.sby bug`.

## Expected behavior

An error message that describes what the issue is. For example, verilator outputs this:

```
verilator --lint-only foo.sv
%Error: foo.sv:6: Found definition of 'signal_id' as a BEGIN but expected a variable
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/bin/verilator_bin --lint-only foo.sv
```

## Actual behavior

The verification run fails because of the syntax error, but the displayed message is not clear enough to identify the problem (` ERROR: Assert ``count_id(cell->name) == 0' failed in kernel/rtlil.cc:1810`):

```
sby -f example.sby bug
SBY 13:57:58 [example_bug] Removing directory 'example_bug'.
SBY 13:57:58 [example_bug] Writing 'example_bug/src/foo.sv'.
SBY 13:57:58 [example_bug] engine_0: smtbmc
SBY 13:57:58 [example_bug] base: starting process "cd example_bug/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 13:57:58 [example_bug] base: ERROR: Assert `count_id(cell->name) == 0' failed in kernel/rtlil.cc:1810.
SBY 13:57:58 [example_bug] base: finished (returncode=1)
SBY 13:57:58 [example_bug] base: job failed. ERROR.
SBY 13:57:58 [example_bug] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 13:57:58 [example_bug] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 13:57:58 [example_bug] DONE (ERROR, rc=16)
SBY 13:57:58 One or more tasks produced a non-zero return code.
```

zachjs, 2021-02-25:
There are at least a few scenarios where this assertion can be encountered. My proposal would be to have a helper in `genrtlil` for producing an as useful as possible error message, ideally including:
- the current kind of high-level construct being added, i.e., something mentioning "assert" would be more useful than just saying "cell" for the conflict in this example
- the file location of the current construct
- the (likely generalized) kind of the already-existing element (wire, memory, cell, or process)
- the source location of the already-existing element (from the RTLIL `src` attribute)

So for this example, I'd like to see something like: ```filename.v:2: ERROR: Cannot add procedural assertion `signal_id' because a wire with same name was already created at filename.v:1!```

whitequark, 2021-02-25:
Sounds good to me.

closed 2021-03-01 by whitequark

jfierro, 2021-03-01:
Wow, that was fast. Thanks!

PR #2615 (2021-03-01): genrtlil: improve name conflict error messaging
Fixes #2612