YosysHQ/yosys #2613: No support for Verilog-2001 non-ANSI port aliases

thoughtpolice, 2021-02-25:
## Steps to reproduce the issue

Given the following (a minimized circuit that was spit out by a compiler), try `read_verilog t.v`

```verilog
module t(.v(t$IO));
  inout  t$IO;
  wire t$IO;

  TriState #(.width(32'd1)) t(.I(1'b0), .OE(1'd0), .O(), .IO(t$IO));
endmodule

module TriState
    ( inout IO
    , output O
    , input OE
    , input I
    );
  assign IO = (OE) ? I : { width { 1'bz } };
  assign O  = IO;
endmodule
```

## Expected behavior

The `v` port for module `t` should point to the `t$IO` net; in effect something like:

```verilog
  assign v = t$IO;
```

More importantly, multiple aliases to the same net should be supported, as in the example from SO:

```verilog
module linked_io ( .io_A(io), .io_B(io), i, c );
  inout io;
  input i, c;
  assign io = c ? i : 1'bz;
endmodule
```

## Actual behavior

Yosys simply errors during parsing module `t`: `t.v:1: ERROR: syntax error, unexpected TOK_ID, expecting '.'`. This is actually a Verilog 2001 feature, apparently(!), which is intended to allow you to alias two ports to point to the same net. Apparently, this is one of the only ways to e.g. drive an `inout` from another `inout`. See https://stackoverflow.com/a/47293378 as well (some tools apparently support `tran` as well, for this case.)

For details, see IEEE-1364-2001 section 12.3.3 `Port declarations`.

## Workarounds

If you only need one IO connection to refer to that net, you can obviously use a `wire`. For multiple I/O aliases to the same net, there is no known workaround I am aware of?

zachjs, 2021-02-28:
The `list_of_ports` style allows for even more than just aliasing. Some highlights:
* A `port_reference` can also be a bit or a slice of another port
* Concatenations of `port_reference`s are allowed, i.e., `` module example(.foo({bar[0:0], bar[1:1]}), bar); ``
* The port names are optional; you can make modules with ports which can only be bound positionally
* The `port_reference` is optional; some tools treat this as forcing instances to leave a port unconnected
* You can omit both of the above simultaneously, i.e., `` module example(, bar); ``

I can imagine how we'd go about adding support for this feature for inputs and outputs. It would entail some restructuring of the module header parsing, but otherwise would basically create synthetic wires and connections in the obvious way.

However, it's not obvious to me how this would be accomplished for an `inout`. Detecting that an `inout` has been referenced in this way seems straightforward, but I'm not sure if there is some reasonable way to represent this in RTLIL. Perhaps someone who understands these mechanics better can clarify?

zachjs, 2021-03-01:
A request for the concatenation feature was actually made back in #860.

thoughtpolice, 2021-03-02:
> You can omit both of the above simultaneously, i.e., module example(, bar);

This has just done an incredible amount of psychic damage to me. My psyche...

hansfbaier, 2022-08-03:
Would the pull request for #3334 be a fix for that

thoughtpolice, 2022-08-05:
@hansfbaier I saw this PR from you a little while ago (last week?), got excited, but forgot to mention this issue might be relevant to your interests &mdash; **thank you** for finding this, commenting, reminding me! Yes, I think that PR might be enough to fix this, though I'd need to test it.

The actual motivation in this case is the ability to parse Bluespec-generated code; aside from this, there are two distinct modules from the Bluespec source code that also need this feature: **[InoutConnect](https://github.com/B-Lang-org/bsc/blob/main/src/Verilog/InoutConnect.v)**, and **[ProbeHook.v](https://github.com/B-Lang-org/bsc/blob/main/src/Verilog/ProbeHook.v)**.

The above test case in my original comment should be self contained, but has no test vectors. Ideally, all three of these need to pass successfully. I can definitely try your branch this weekend perhaps and see what it does. I think we should probably add these 3 files all as test cases for it. (Ideally we should add them *no matter what* and just mark them as expected failures, and then supporting this feature would require marking those tests as expected successes &mdash; though I don't know how the testsuite works right now and if it supports that.)