YosysHQ/yosys #2614: Latency 5+ self unproven circuit

drom, 2021-02-25:
## Steps to reproduce the issue

The following Verilog have: `ERROR: Found 4 unproven $equiv cells in 'equiv_status -assert'.` with itself.

```verilog
module top_mod(
  input        clock,
  input        reset,
  output [3:0] tmp30,
  input  [3:0] inp_i
);
  reg [3:0] _tmp4;
  reg [3:0] tmp7;
  reg [3:0] tmp8;
  reg [3:0] tmp10;
  reg [3:0] tmp19;
  assign tmp30 = tmp19;
  always @(posedge clock) begin
    if (reset) begin
      _tmp4 <= 4'h0;
      tmp7 <= 4'h0;
      tmp8 <= 4'h0;
      tmp10 <= 4'h0;
      tmp19 <= 4'h0;
    end else begin
      _tmp4 <= inp_i;
      tmp7 <= _tmp4;
      tmp8 <= tmp7;
      tmp10 <= tmp8;
      tmp19 <= tmp10;
    end
  end
endmodule
```

## Expected behavior

to be equivalent with itself

## Actual behavior

not equivalent to itself

mwkmwkmwk, 2021-02-26:
Please include full reproduction information, including the yosys commands used.

drom, 2021-02-26:
added bash script above

mwkmwkmwk, 2021-02-26:
That's by design (and a general limitation of the formal methods used here) — this design requires 5 induction steps to prove as equivalent, while the default for `equiv_induct` is 4 steps.  You can get it to work by running `equiv_induct -seq 5` (or more).

closed 2021-02-26 by mwkmwkmwk