YosysHQ/yosys #2616: CXXRTL: failed backend invocation leaves empty output file

whitequark, 2021-02-27:
## Steps to reproduce the issue

```
cat >repro.v <<END
module a; endmodule
module top; a a(); endmodule
END
yosys repro.v -o repro.cc
```

## Expected behavior

`repro.cc` is either not created, or removed after being created, to play nice with build systems.

## Actual behavior

Empty `repro.cc` is created, breaking downstream build if it is the sole target in the dependency graph.

## Notes

This might be affecting more backends than CXXRTL, CXXRTL is just particularly picky about the input.

arcturusannamalai, 2023-05-25:
this happens because of https://github.com/YosysHQ/yosys/issues/3762

whitequark, 2023-05-25:
I don't actually see how that's caused by not having `post_execute()`.