YosysHQ/yosys #2648: ice40_dsp, extend_u0 and unextend incorrectly handle sign bits

sylefeb, 2021-03-09:
## Steps to reproduce the issue

On yosys 26e01a67db4135196b1d25ed89e9e6ceb536f4e3.

Using the three files in the gist https://gist.github.com/sylefeb/447c22fa39b25e7056ba4e4ecae92d7b
one can see that the results differ whether using the `-dsp` option or not. Without `-dsp` the result is:
`result = 117 (should be 117)`
With -dsp the result is:
`result = 5 (should be 117)`

## Expected behavior

The result with and without `-dsp` should match, the result is correct in this case without `-dsp`.

## Actual behavior

The result differs (5 vs 117).

## Analysis

It appears the problem stems from the combined behavior of `unextend` and `extend_u0` in `ice40_dsp.pgm` and `ice40_dsp.cc`. My understanding is that: 
- `unextend` removes leading bits from signals if they are the same (wire or const), including the bit that may represent a change of sign through e.g. a two's complement on an unsigned wire (and including the sign bit of a const signal, regardless of signedness).
- `extend_u0` assumes leading bits are always zeros when padding back an unsigned signal

## Suggested fix

I am not sure enough to propose this as a PR, but this works on my side in all tests so far:
- always preserve what represents a sign bit (including on unsigned signals) in `unextend`
- always give `extend_u0` a 'signed' signal (second parameter always true), such that the MSB is replicated during extension

This may be too conservative however.

Patch: https://github.com/sylefeb/fpga-binutils/blob/master/patches/yosys_patch_ice40_dsp.diff

## Remarks

- A similar issue is also fixed by the patch, which affects signed const signals loosing their sign bit in `unextend`.
- Other platforms (xilinx dsp) may be affected by a similar issue as they also use `unextend`.

emeb, 2021-03-22:
I also had an issue with connection of signed values to the SB_MAC16 cell and this patch appears to have corrected it.

bit-hack, 2021-09-06:
I think I have also encountered this issue myself.  I have observed incorrect and differing behavior when using the `-dsp` option to without it.  I compiled this code:
```v
module mdac12x8 (
    input                clk,
    input  signed [11:0] iVoice,
    input         [ 7:0] iEnv,
    output signed [15:0] oOut
);
  wire signed [15:0] lhs = { iVoice, 4'h0 };
  wire signed [15:0] rhs = { 8'h0, iEnv };

  reg signed [31:0] product;
  assign oOut = product[23:8];
  always @(posedge clk) begin
    product <= lhs * rhs;
  end
endmodule
```

I compiled it as follows:
```
yosys -f verilog -l log.yslog -p 'read_verilog input.v; synth_ice40 -dsp -json out.json; write_verilog output.v'
```

In the post synthesis output we can see that:
The upper 8 bits of the `B` input (zeros) have been discarded, and then the lower byte has been sign extended.

```
  ) oOut_SB_MAC16_O (
    .A({ iVoice[11], iVoice[11], iVoice[11], iVoice[11], iVoice }),
   ...
    .B({ iEnv[7], iEnv[7], iEnv[7], iEnv[7], iEnv[7], iEnv[7], iEnv[7], iEnv[7], iEnv }),
 ```

I think this accounts for the different and incorrect behavior I have seen.