YosysHQ/yosys #2652: how to support Intel Stratix10

gnleaf, 2021-03-11:
I want to use yosys on Intel Stratix 10.
so I must support ALM （version of Stratix 10 ）.
Any good suggestions?

Ravenslofty, 2021-03-17:
I haven't yet come across a Stratix 10 user, which is why there's presently no support for it. I could probably add something resembling synthesis support, but it would be highly experimental because I simply couldn't run tests.

thoughtpolice, 2021-03-18:
I do have an S10 device available but I admittedly have not spent much time investigating support for using it with Yosys, mainly because I'd have to re-acquire a Quartus license for full bitstream testing, and the Yosys Intel support has always seemed rather preliminary. I think many of the primitives for the ALM family of devices (e.g. `dffeas`) are still the same though, and even the basic ALM/MLAB organization looks the same between Arria10 and Stratix10 &mdash; so the basics probably look a lot like what is already there. In fact it might mostly work already, but I'd have to test it and get it through Modelsim I guess...

Ravenslofty, 2021-03-18:
(as a note, I tend to discourage use of `dffeas` because the primitive does not reflect actual hardware support; instead I plan to switch to `<family>_ff` cells instead)

gnleaf, 2021-03-31:
> I do have an S10 device available but I admittedly have not spent much time investigating support for using it with Yosys, mainly because I'd have to re-acquire a Quartus license for full bitstream testing, and the Yosys Intel support has always seemed rather preliminary. I think many of the primitives for the ALM family of devices (e.g. ) are still the same though, and even the basic ALM/MLAB organization looks the same between Arria10 and Stratix10 — so the basics probably look a lot like what is already there. In fact it might mostly work already, but I'd have to test it and get it through Modelsim I guess...`dffeas`

I have a S10 device available and the license of quartus, so I can test it