YosysHQ/yosys #2656: PROC_ARST misinterprets a clock as an async reset

cyrozap, 2021-03-12:
## Steps to reproduce the issue

Create a file called `testcase.v` with the following contents:

```verilog
module testcase (
   input correct_clk,
   input reset_q,
   input do0_d,
   output do0,
   );

   reg do0_q;

   assign do0 = do0_q;

   begin : read_data_latched_true
      always @(posedge correct_clk)
      begin: read_data_latches
         if (correct_clk == 1'b1)
         begin
            if (reset_q == 1'b1)
            begin
               do0_q <= 1'b0;
            end
            else
            begin
               do0_q <= do0_d;
            end
         end
      end
   end
endmodule
```

(This is a simplified test case based on [this file](https://github.com/openpower-cores/a2o/blob/07e252685a9a163dc137af06dbedb707d4f0b43b/rel/src/verilog/trilib/tri_iuq_cpl_arr.v) from [A2O](https://github.com/openpower-cores/a2o).)

Then run `yosys -p 'read_verilog testcase.v; proc'`.

## Expected behavior

`correct_clk` is interpreted as a clock signal, not an asynchronous reset.

The [A2O](https://github.com/openpower-cores/a2o) core is normally synthesized with Vivado, so I would think if Vivado can handle this without any problems, it should be possible for Yosys to do so as well.

## Actual behavior

The following error gets printed:

```
2.5. Executing PROC_ARST pass (detect async resets in processes).
Found VHDL-style edge-trigger \correct_clk in `\testcase.$proc$testcase.v:13$1'.
ERROR: Async reset \correct_clk yields non-constant value 1'm for signal \do0_q.
```

## Workaround

Remove the `if (correct_clk == 1'b1)`, like so:

```verilog
module testcase2 (
   input correct_clk,
   input reset_q,
   input do0_d,
   output do0,
   );

   reg do0_q;

   assign do0 = do0_q;

   begin : read_data_latched_true
      always @(posedge correct_clk)
      begin: read_data_latches
         if (reset_q == 1'b1)
         begin
            do0_q <= 1'b0;
         end
         else
         begin
            do0_q <= do0_d;
         end
      end
   end
endmodule
```

mwkmwkmwk, 2021-03-12:
I'm going to fix the error, but do note that writing code like this (looking at the value of the clock signal within a sync process) is a supremely bad idea that gains you nothing and is sure to result in a simulation-synthesis mismatch — simulation will always observe the `correct_clk` value as 1 due to the posedge trigger, while the synthesized circuit will either always observe 0 (for big enough combinatorial delays and/or negative FF hold time) or go metastable

mwkmwkmwk, 2021-03-12:
After some more deliberation, we decided to implement a special case for this pattern, since some other synthesis tools do that, and it appears in real-life code.  However, it remains a bad idea.

PR #2657 (2021-03-15): proc_arst: Add special-casing of clock signal in conditionals.
The already-existing special case for conditionals on clock has been
remade as follows:

- now triggered for the last remaining edge trigger after all others
  have been converted to async reset, not just when there is only one
  sync rule in the first place
- does not require all contained assignments to be constant, as opposed
  to a reset conditional — merely const-folds the condition

In addition, the code has been refactored a bit; as a bonus, the
priority order of async resets found is now preserved in resulting sync
rule ord [... cut]
commit a55bf637 (2021-03-15): proc_arst: Add special-casing of clock signal in conditionals.

The already-existing special case for conditionals on clock has been
remade as follows:

- now triggered for the last remaining edge trigger after all others
  have been converted to async reset, not just when there is only one
  sync rule in the first place
- does not require all contained assignments to be constant, as opposed
  to a reset conditional — merely const-folds the condition

In addition, the code has been refactored a bit; as a bonus, the
priority order of async resets found is now preserved in resulting sync
rule or [... cut]
closed 2021-03-15 by mwkmwkmwk