YosysHQ/yosys #2661: Error occurs when building yosys-master

TheRitzCarlton, 2021-03-15:
When I type in “make”, I get:

bash: tclsh: command not found
bash: tclsh: command not found
[  0%] Building kernel/driver.o
In file included from kernel/driver.cc:20:
./kernel/yosys.h:81:12: fatal error: 'tcl.h' file not found
#  include <tcl.h>
           ^~~~~~~
1 error generated.
Makefile:708: recipe for target 'kernel/driver.o' failed
make: *** [kernel/driver.o] Error 1

How to solve that? I saw the introduction says that tcl is optional.THX

Xiretza, 2021-03-15:
Have you tried setting `ENABLE_TCL=0` either as a make command line option or in `Makefile.conf`?

gatecat, 2021-03-15:
Or you can ensure that tcl and its development library is installed (the details for this will depend on your OS.)

TheRitzCarlton, 2021-03-15:
But now I need a version which supports TCL execution as I need to run it together with nextpnr-generic. 
Is there an approach to make the yosys version without disabling TCL？

TheRitzCarlton, 2021-03-15:
> Or you can ensure that tcl and its development library is installed (the details for this will depend on your OS.)

I‘ll try it. THX

zachjs, 2021-03-17:
@TheRitzCarlton Has this issue been resolved?

closed 2021-03-17 by zachjs