YosysHQ/yosys #2662: How can a user black-box a module when using the smt2 backend?

rswarbrick, 2021-03-15:
I'm working with Symbiyosys, but I think this question is really about Yosys itself. If I want to work with a design that contains memories, I might not actually care about the exact behaviour of the memories - maybe I'm just interested in control signals for surrounding blocks.

The "obvious" thing to do would be to black-box the RTL module that represents the memory. Symbiyosys doesn't allow this, running a `hierarchy -simcheck` at the end of the script. It turns out that this is for a good reason! If you comment that check out, it seems that the SMT backend spits out something involving the black-boxed module that the solver doesn't like (I see something like `boolector: <stdin>:41:29: invalid sort '|$paramod/prim_rom_adv/Width=40/Depth=4096/MemInitFile=1'0_s|'.`).

Is there a proper way to black-box a module within the design?

nakengelhardt, 2021-04-19:
The `cutpoint` command can remove an element from the design and drive the wires with `$anyseq` instead, for formal blackboxing. I just expanded [the readme for the example using cutpoint](https://github.com/YosysHQ/SymbiYosys/tree/master/docs/examples/abstract) in SymbiYosys a bit.

rswarbrick, 2021-04-19:
Ah, brill! Thanks for that. If the thing I'm trying to black-box is big, will I also want to flush it from the design somehow (to avoid the solver reasoning about it)? Reading the documentation quickly, it looks like it will stay in the design, but "waggling in the breeze" :-) Or have I misunderstood something?

rswarbrick, 2021-04-26:
Oh, great. I'll try it out, thanks!

rswarbrick, 2021-06-15:
Oops, I've just realised I didn't reply to this. It works perfectly, thanks.

closed 2021-06-15 by rswarbrick