YosysHQ/yosys #2667: Processes in whitebox causes synth_ecp5 JSON backend to fail

gatecat, 2021-03-16:
Run:

```
yosys -p "add -mod top; synth_ecp5 -json /dev/null"
```

as the ECP5 distributed RAM whiteboxes contain processes but nothing elaborates those processes; the JSON backend is now failing:

```
ERROR: Module DPR16X4C contains processes, which are not supported by JSON backend.
```

A workaround would be to unwhitebox those cells as I don't think there is much useful optimisation that can be done anyway, but I suspect this issue may well affect other situations involving boxes in processes as well.

cc @edbordin

mwkmwkmwk, 2021-03-16:
Would running `blackbox` command on all whitebox modules at the end of the synthesis flow solve this problem?

gatecat, 2021-03-16:
Yes, that seems like a better way to deal with this

PR #2679 (2021-03-17): Blackbox all whiteboxes after synthesis
This prevents issues like processes in whiteboxes triggering an error in the JSON backend. Fixes #2667
closed 2021-03-17 by gatecat