YosysHQ/yosys #2677: SV struct as type.

Kmanfi, 2021-03-17:
## Steps to reproduce the issue

bug1.sv
```
typedef struct packed
{
    logic unsigned                       [4:0]   p_value;
} ch_struct_t;

typedef struct packed
{
    ch_struct_t  [7:0]  channels;
} ch_set_t;

```
`read_verilog -sv bug1.sv`

## Expected behavior
Read wo error.

## Actual behavior

`bug1.sv:8: ERROR: syntax error, unexpected '[', expecting TOK_ID`

zachjs, 2021-03-17:
Arrays of structs are not currently handled by the frontend. You will also find that structure literals are not supported. It will likely be some time before these features can be added. If you are looking for an immediate workaround, you may be interested in https://github.com/zachjs/sv2v.

Kmanfi, 2023-02-09:
Status of this feature?

oharboe, 2023-07-25:
@seldridge This would be useful with [LLVM CIRCT](https://circt.llvm.org/) that uses packed arrays by default.

Another example:

```verilog
module Foo(
  input  [63:0]  a_0,
                 a_1,
                 a_2,
                 a_3,
  input  [1:0]   b,
  output [127:0] c
);

  wire [3:0][63:0] _GEN = {{a_3}, {a_2}, {a_1}, {a_0}};
  assign c = {2{_GEN[b]}};
endmodule
```

No complaints from Verilator:

```
$ ~/verilator/install/v5.006/verilator/bin/verilator --lint-only Foo.v
$ ../tools/install/yosys/bin/yosys
[deleted]
 Yosys 0.13+15 (git sha1 bc027b2ca, gcc 11.3.0-1ubuntu1~22.04 -fPIC -Os)
[deleted]
yosys> read_verilog Foo.v
1. Executing Verilog-2005 frontend: Foo.v
Parsing Verilog input from `Foo.v' to AST representation.
Foo.v:10: ERROR: syntax error, unexpected '[', expecting TOK_ID or '#'
```

daglem, 2024-02-03:
> @seldridge This would be useful with [LLVM CIRCT](https://circt.llvm.org/) that uses packed arrays by default.
> 
> Another example:
> 

@oharboe Your example requires support for another feature; packed multidimensional arrays. This is implemented in #4100.