YosysHQ/yosys #2684: SV function

Kmanfi, 2021-03-18:
## Steps to reproduce the issue

bug2.sv
```
package in_types_pkg;
typedef struct packed
{
    logic unsigned  [1:0]  m_v;
} in_struct_t;
endpackage

package out_types_pkg;
typedef struct packed
{
    logic unsigned  [1:0]  m_v;
} out_struct_t;
endpackage

function automatic out_types_pkg::out_struct_t e_s (
    input in_types_pkg::in_struct_t din
);
    out_types_pkg::out_struct_t dout;
    dout.m_v = din.m_v;
    return dout;
endfunction
```
`read_verilog -sv bug2.sv`
## Expected behavior

Read wo error.
## Actual behavior

bug2.sv:16: ERROR: syntax error, unexpected TOK_PKG_USER_TYPE

zachjs, 2021-03-18:
I have a PR incoming to add support for typenames as function return types. This was pretty easy to add.

Supporting `return` as used here could be straightforward, but the general case is far more complex. Supporting the jump keywords would be a significant undertaking. https://github.com/zachjs/sv2v does handle this already, and so may be a viable workaround.

Kmanfi, 2021-03-23:
So, "return" as it's in example bug2.sv is not supported yet?

zachjs, 2021-03-23:
> So, "return" as it's in example bug2.sv is not supported yet?

That's correct.