YosysHQ/yosys #2687: Allow the possibility of a capture mode

modwizcode, 2021-03-18:
Since in the process of debugging issues within yosys it might only be reproducible as part of a potentially multilayer flow of upstream elements it seems prudent to implement some kind of argument or setting (probably a command line argument) to allow turning yosys into a dummy mode where it will simply record to a file enough state to ingest the same form and reproduce the response to the original state. 

Ideally this mode could be coupled with actually working as intended in order to not trip up processes that would fail if artifacts are not produced.

Hopefully this is both not a totally stupid idea and also isn't already a feature that I've missed somehow, but it would be extremely useful. I'm sure other solutions could exist but I can't imagine them working nearly as well as giving yosys this functionality internally, internal support would allow capturing the complete set of both input files and transformation at runtime such that even ephemeral input data was capturable.

For an example of what I can imagine would require this, imagine a system where code is developed locally and yosys runs on a build server. Or some kind of parallelized build (or at least validation) where independent ip blocks are generated/verified in parallel. These kinds of configs would likely be totally dynamic and extremely difficult to reproduce the exact sequence of steps leading up to the issue.

clairexen, 2021-03-22:
I have problems understanding what exactly you are asking for.

But `yosys -E outfile` will generate a file with all dependencies (files read and generated) in Makefile syntax.

Another useful pointer might be the `bugpoint` command within Yosys that can be useful for pinpointing a bug within a Yosys flow.

closed 2021-03-22 by clairexen

modwizcode, 2021-03-22:
Yeah I think I was a bit unclear. Basically I'm imagining someone has a large design split into a considerable number of files, and they go to submit a bug, but the origin of the bug is unclear. Also consider that several of my inputs to yosys are automatically generated at synthesis time.

Depending on project structure, actually producing an equivalent run for debugging might be difficult because it might require running the exact same set of inputs with the exact same sequence of yosys commands. At least until someone can dig into the problem and find the failure.

However if use of yosys is scripted or automated in some way (which I find to be likely), actually capturing both the *contents* of the set of files read and the execution flow (read: commands passed to yosys and their arguments), might be difficult. Transient state such as build time generated files especially complicate the problem.

The suggestion here is that really no point is well suited in the process for logging the inputs and outputs except yosys itself. Especially if you imagine possible designs where yosys reads files from a pipe or something like that.

Ravenslofty, 2021-03-23:
One issue is that you can't fully serialise Yosys state in a reproducible manner; parametric modules will not serialise because elaboration of modules depends heavily on the AST, which isn't serialised at all.

modwizcode, 2021-03-23:
@Ravenslofty Are you saying that's an issue for implementation of this or saying that even if someone implemented what I suggest that's not enough? I'm not entirely sure where the issue would occur that you suggest.

Are you saying that even if you give the same version of yosys the same input files and the same commands, the elaboration wouldn't match? That seems odd. Regardless, the goal here isn't to truly serialize intermediary state, but to provide an avenue towards easing the ability for users (or developers) of yosys itself to capture their failing inputs across a wider range of setups, pursuant to their submission of a bug report that can be reproduced (and hopefully minimized).

For example, if I use a tool like nMigen to not only handle generating the inputs to yosys but also take advantage of it's build system functionality combined with a specific fpga platform to go all the way to final bitstream, I would have to wander into the source of nMigen itself and manually expand the templated commands it runs, not to mention independently collecting the intermediate generated input HDL.

clairexen, 2021-03-23:
`yosys -E` captures the input file names, including the name of the script. Not executing the script would be counterproductive because you need to execute the script to know which files yosys reads.

Ravenslofty, 2021-03-24:
@modwizcode I'm saying that if you dump a failing input to RTLIL and then try to run Yosys again on that RTLIL, if the input has parametric modules, the second Yosys run will fail because it does not have the AST necessary to elaborate those modules.

clairexen, 2021-03-29:
Copy&paste from today's dev jour fixe minutes:

This really looks like it should be done in whatever generates the input files for Yosys. If for no other reason than that no other tool will support this, so the moment they run into issues with any other tool, they would need a feature to store away the generated files anyway.

Also, it’s a lot of nontrivial work for something that does not seem to be a big issue that many people run into.