YosysHQ/yosys #2690: Inconsistent behavior of signed shift inside ternary operation

markzakharov, 2021-03-19:
## Steps to reproduce the issue

    `    
    module randomverilog(
    input clock,
    input reset,
    input [0:0] io_a0, io_a1, io_a2, io_a3, io_a4,

    output [0:0] io_y);

    // These two are NOT equivalent
    //assign io_y= ((((io_a4))?(io_a0):(~ (($signed(io_a3) >>> io_a2)) )));
    //assign io_y= ((((io_a4))?(io_a0):(~ {($signed(io_a3) >>> io_a2)} )));

    // These two are equivalent
    //assign io_y= ~(($signed(io_a3) >>> io_a2));
    //assign io_y= ~{($signed(io_a3) >>> io_a2)};

    // Simple Test
    assume property (!io_a4);  //mux select
    assume property (io_a2);   //sets shift
    assume property (io_a3);   //makes sure a3 has 1, should ideally maintain 1 after signed shift
    assert property (!io_y);    //inversion makes io_y==0 ideally
    endmodule 
    `

## Expected behavior

When selected in the ternary, a signed right shift should occur for both parenthesis and braces as the inputs and the resulting width of the concatenation is 1 bit. In this case the signed operation happens within braces, which can also be interpreted as parenthesis when there is only a single element.

## Actual behavior

The first assign statement does not perform a signed right shift. It only works when it is interpreted as a concatenation. Though once outside of the ternary operation, the right shift performs as expected. Running the assume/assert should yield the same result, but only the bottom 3 assign statements agree with the assert. Somehow being in parenthesis in the ternary operation is forcing the shift to be unsigned.

Thanks to @renau for isolating the issue

zachjs, 2021-03-19:
I believe this is not a bug, but a classic case of Verilog's expression evaluation rules producing surprising results.

I have a reduced version of the above example which exhibits the same discrepancy, but is a bit easier to reason about.
```Verilog
module top(
	output wire out1, out2
);
	wire inp1, inp2, inp3, inp4;
	assign inp1 = 1'b0;
  	assign inp2 = 1'b1;
  	assign inp3 = 1'b1;
  	assign inp4 = 1'b0;

	assign out1 = inp4 ? inp1 : ($signed(inp3) >>> inp2);
	assign out2 = inp4 ? inp1 : {$signed(inp3) >>> inp2};

	initial #1 $display("%b %b %b", out1, out2, out1 === out2);
endmodule

```

For every simulator I can get my hands on (commercial or otherwise) _except for one_, this example produces:
```
0 1 0
```

My reading of the LRM agrees with the majority of simulators here, and with Yosys's current behavior.

From IEEE 1800-2017 Section 11.4.10 - Shift operators: (emphasis mine)
> The arithmetic right shift shall fill the vacated bit positions with zeros if the **result type** is unsigned. It shall fill the vacated bit positions with the value of the most significant (i.e., sign) bit of the left operand if the **result type** is signed.

Additionally, Table 11-21 explains that all operands of a concatenation are self-determined expressions.

Because `inp1` is unsigned, the result of the ternary expression is unsigned, and hence the third operand is unsigned in both cases. In the assignment of `out1`, `($signed(inp3) >>> inp2)` is therefore unsigned, and so must fill with zeroes when shifted. However, within `{$signed(inp3) >>> inp2}`, `$signed(inp3) >>> inp2` is self-determined, and so takes on the signedness of `$signed(inp3)`, filling with the sign bit when shifted.

zachjs, 2021-03-24:
I hope my above comment demonstrates that the existing behavior is in accordance with the LRM. Given there hasn't been further activity since then, I am closing this issue. Please feel free to reopen or comment if necessary!

closed 2021-03-24 by zachjs