YosysHQ/yosys #2717: unpacked array in ports

mehdimolum, 2021-03-30:
Hi, I am wondering if there is any plan to support multi-dimensional arrays (packed/unpacked) in ports. yosys already supports multidimensional arrays within a module but how involved is it to extend the feature to ports as well?

jgaztelu, 2021-05-13:
This would be much appreciated

zachjs, 2021-07-09:
I believe the work to get this implemented might be significant. In the meantime, you may be interested in trying out https://github.com/zachjs/sv2v, which is already capable of handling these features.

julian1, 2023-10-26:
Is there a workaround to this?
My use-case is a configurable register bank, where registers all the same length can be indexed, instead of defined on a per-register basis. But it needs inputs and outputs for the interface.

I tried to use a generate to assign but get an error,

https://old.reddit.com/r/yosys/comments/44d7v6/arrays_as_inputs_to_modules/

register_set_01.v:78: ERROR: Left hand side of 1st expression of generate for-loop is not a genvar!

JDuchniewicz, 2025-01-08:
Confirming it is still open - at least with th ecommit SHA `80ba43d26`. AFAIK you can flatten the array when passing to the module and then de-flatten it inside.