YosysHQ/yosys #2723: Process crashed and killed when attempting to read verilog

CHI-MING-LEE, 2021-04-08:
## Steps to reproduce the issue

The attached was the EPFL Combinational Benchmark Suite consisting of arithmetic, random control and 3 MtM benchmarks.
When attempting to read the verilog into yosys **after yosys-abc transformations** on sixteen.v, yosys consumed about 120G memory and crashed. Is there any way to handle this problem? The commands for reproduce are listed below:

```shell=
yosys-abc -c read osu018_stdcells.lib; read sixteen.v; strash; refactor; rewrite; balance; write sixteen_opt.v; map -D 100000; topo; stime;
yosys -QT -p read sixteen_opt.v
```

[osu018_stdcells](https://vlsiarch.ecen.okstate.edu/flows/MOSIS_SCMOS/osu_soc_v2.5/cadence/lib/tsmc018/signalstorm/osu018_stdcells.lib)
[The EPFL Combinational Benchmark Suite](https://zenodo.org/record/2572934#.YG6R0WgzaUl)

Please unzip the file and find the sixteen.v under EPFLfull\MtM\sixteen\verilog

clairexen, 2021-04-12:
It looks like you want to use something like BLIF instead of Verilog as exchange format between ABC and Yosys. Beyond that I can't say anything about the expected size of the circuit in-memory, since you have not shared any information on the size of the `sixteen_mapped.v` file, and the expected number of cells in the netlist is describes.

(Btw, it looks like all the quotes are missing from the code you posted.)

CHI-MING-LEE, 2021-04-13:
Thank you for your reply @clairexen.
I can confirm BLIF format works. I can `read_verilog sixteen.v` from ABC first and write as BLIF for Yosys to use. However, when I `read_verilog sixteen.v`, Yosys still crashed due to exhausted memory (~120G). What's the difference between ABC and Yosys on `read_verilog` command? 

[The EPFL Combinational Benchmark Suite](https://zenodo.org/record/2572934#.YHVGSGgzaUm)
After unzipping [this](https://zenodo.org/record/2572934/files/EPFL_complete.tar.gz?download=1), the design `sixteen.v` is under EPFLfull\MtM\sixteen\verilog

nakengelhardt, 2021-04-14:
The `read_verilog` frontend is not designed for synthesized netlists, it creates many data structures used for high-level analysis and transformations of the design and for traceability, so every single cell and wire will have a significant amount of data attached to it. (In particular, for each element there is detailed source location information stored in a particularly inefficient format, which we suspect causes a lot of the bloat.) The front-ends for formats geared towards exchange with other low-level tools like abc or nextpnr have less overhead and are recommended when working with large and/or fine-grain netlists.

nakengelhardt, 2021-04-26:
We tried a few things but couldn't find a simple fix to reduce memory usage of `read_verilog` short of rewriting the entire frontend (which is already on the nebulous plan for the distant future), so I'm closing this issue now.

closed 2021-04-26 by nakengelhardt