YosysHQ/yosys #2725: ENABLE_NDEBUG = 1 fail.

Kmanfi, 2021-04-09:
## Steps to reproduce the issue
Current head compile and tests passed but if I only change 
ENABLE_NDEBUG := 1
it will fail.

This works.
```
ENABLE_NDEBUG := 0
 ./yosys --version
Yosys 0.9+4052 (git sha1 0ccc7229c, clang 11.0.0 -fPIC -Os)
```

```
clang --version
Apple clang version 11.0.0 (clang-1100.0.33.16)
Target: x86_64-apple-darwin20.3.0
Thread model: posix

```
## Expected behavior

make test 
Passed "make test".

## Actual behavior
This doesn't work.
(yes I cut my yosys-abc path away on this post .....)
```
ENABLE_NDEBUG := 1

Passed logic.ys
Passed lutram.ys
sh: line 1: 95834 Abort trap: 6           ...../yosys-abc -s -f /tmp/yosys-abc-mcIbVt/abc.script 2>&1
ERROR: ABC: execution of command "...../yosys-abc -s -f /tmp/yosys-abc-mcIbVt/abc.script 2>&1" failed: return code 134.
make[1]: *** [mux.ys] Error 1
make: *** [test] Error 2
```

clairexen, 2021-04-12:
The `/tmp/yosys-abc-mcIbVt/` sidrectory is auto-generated by Yosys and it contains all the input files for ABC. (The directory name is different for each run.)

We can not reproduce the issue on our side. Can you reproduce it again on your side, zip the contents of that directory, and post that here?

Kmanfi, 2021-04-13:
I try to make zip that You can reproduce this case.

clairexen, 2021-04-26:
> I try to make zip that You can reproduce this case.

Hey there.  o/
Still waiting for that zip file.  ;)

mwkmwkmwk, 2021-07-27:
A similar problem (SIGABRT in yosys-abc on mac os) happens semi-regularly with our mac os github actions builder: see https://github.com/YosysHQ/yosys/pull/2892/checks?check_run_id=3172120181 for an example failed run (I recall seeing it a few more times as well, it's not an isolated incident), *and* it seems to be very non-deterministic (I force-pushed the exact same code that failed in this run with just a new commit timestamp and the build succeeded).  No `ENABLE_NDEBUG` changes were involved here, however, so it's unlikely to be the cause (more likely, the general nondeterminism of the bug caused it to happen in `ENABLE_NDEBUG` run for the OP).

mwkmwkmwk, 2021-07-27:
... oh great, github already removed the logs for that run; copying from a still-open tab:

```
cd tests/arch/ice40 && bash run-test.sh ""
Warning: Resizing cell port top.sc6.c0.CI from 32 bits to 1 bits.
Warning: Resizing cell port top.sc6.l0.I3 from 32 bits to 1 bits.
Warning: Resizing cell port top.sc6.lien.I0 from 32 bits to 1 bits.
Warning: Resizing cell port top.sc6.lien.I1 from 32 bits to 1 bits.
Warning: Wire top.\PIN_24 is used but has no driver.
Warning: Wire top.\PIN_23 is used but has no driver.
Warning: Wire top.\PIN_22 is used but has no driver.
Warning: Wire top.\PIN_21 is used but has no driver.
Warning: Wire top.\PIN_20 is used but has no driver.
Passed bug1597.ys
Passed add_sub.ys
Passed bug1626.ys
Passed bug1598.ys
Passed counter.ys
Passed adffs.ys
Passed dffs.ys
Passed bug1644.ys
Passed ice40_dsp.ys
Passed ice40_opt.ys
Passed ice40_wrapcarry.ys
Passed fsm.ys
Passed logic.ys
Passed latches.ys
sh: line 1: 76758 Abort trap: 6           /Users/runner/work/yosys/yosys/yosys-abc -s -f /tmp/yosys-abc-VLg9na/abc.script 2>&1
ERROR: ABC: execution of command "/Users/runner/work/yosys/yosys/yosys-abc -s -f /tmp/yosys-abc-VLg9na/abc.script 2>&1" failed: return code 134.
make[1]: *** [memories.ys] Error 1
make[1]: *** Waiting for unfinished jobs....
Passed macc.ys
Passed dpram.ys
make: *** [test] Error 2
Error: Process completed with exit code 2.
```

mwkmwkmwk, 2021-07-27:
I have reproduced this (or a similar enough issue) on Linux, by compiling abc with ASan; the failing assert seems to be https://github.com/YosysHQ/abc/blob/yosys-experimental/src/opt/lpk/lpkCut.c#L200 which blows up on a perfectly valid pointer with the value of eg. `0x627000000568` in my case.  The failure seems to be deterministic for Linux+ASan, on tests/arch/ice40/memories.ys.

Yes, this issue is literally abc not being 64-bit-clean.

clairexen, 2021-07-27:
So the reason it isn't deterministic without ASan is probably ASLR?

mwkmwkmwk, 2021-07-27:
Without ASan, but with ASLR, your pointers on 64-bit machines have 1/65536 probability of containing 0 in bits 16-31 and triggering this assert, hence nondeterministic failure

With ASan, the allocated arenas tend to get very large alignments, often giving you 4GB-aligned objects, resulting in a near-deterministic boom

I cannot figure out why we're not hitting this more often on Linux though, perhaps some quirk of the allocator avoids addresses with 0s in these bits

mwkmwkmwk, 2021-07-27:
Alright, I managed to reproduce it on a default unmodified Linux build, happens roughly one in 20000 iterations of an abc script extracted from `tests/arch/ice40/memories.ys`

my completely wild and unconfirmed guess why it affects macOS more is that its allocator spreads allocations across the virtual address space more, giving you more varied addresses and more opportunities to hit the assert, while Linux keeps all its allocations in one place so either you get the heap somewhere close to a 0-ful address in a given run or not

mwkmwkmwk, 2021-08-02:
[yabc-bug.zip](https://github.com/YosysHQ/yosys/files/6917604/yabc-bug.zip)