YosysHQ/yosys #2729: BLIF frontend does not honor reversed order of port bit indices

mkurc-ant, 2021-04-13:
When a cell in a Verilog library has a wide port with reversed bit indices (eg. `input A[0:3]` instead of `input A[3:0]`) then performing a round trip of a Verilog design through BLIF and back to Verilog will effectively reverse the port indices.

Consider the following cell library. The second one has reversed indices of the `in` port:
```
(* blackbox *)
module my_cell_normal (
    input  wire [3:0] in,
    output wire       out
);
endmodule

(* blackbox *)
module my_cell_reversed (
    input  wire [0:3] in,
    output wire       out
);
endmodule
```

Then in a design those cells are instantiated and individual signal vectors are connected to their input ports:
```
module top(
    input  wire      I0,
    input  wire      I1,
    input  wire      I2,
    input  wire      I3,
    output wire      out_a,
    output wire      out_b
);
    my_cell_normal normal_inst (
        .in  ({I3, I2, I1, I0}),
        .out (out_a)
    );
    my_cell_reversed reversed_inst (
        .in  ({I3, I2, I1, I0}),
        .out (out_b)
    );
endmodule
```

After writing this design to a BLIF file and loading it back (and writing verilog for comparison) the signal ordering should be the same while it isn't:
```
  my_cell_normal normal_inst (
    .in({ I3, I2, I1, I0 }),
    .out(out_a)
  );
  my_cell_reversed reversed_inst (
    .in({ I0, I1, I2, I3 }),
    .out(out_b)
  );
```

even though the order is correct (in the sense of the cell definition) in the BLIF:
```
.subckt my_cell_normal in[0]=I0 in[1]=I1 in[2]=I2 in[3]=I3 out=out_a
.cname normal_inst
.subckt my_cell_reversed in[3]=I0 in[2]=I1 in[1]=I2 in[0]=I3 out=out_b
.cname reversed_inst
```

## Steps to reproduce the issue

Use the files from the attached archive file. Invoke Yosys for each of the scripts present there:
 - `yosys -s script1.v`
 - `yosys -s script2.v`

Compare order of signals connected to the port `in` of each instanced cell. The order for the cell `reversed_inst`  in `my_design.eblif.v` should be reversed when compared to the original design `my_design.v`.

## Expected behavior

The BLIF frontend honors ordering of port bit indices

## Actual behavior

The BLIF frontend does not respect port bit ordering defined in a cell library.

clairexen, 2021-04-19:
> Use the files from the attached archive file.

There's no attached archive file. Please attach it.

mkurc-ant, 2021-04-19:
I've re-uploaded the file and attached a link to it to the issue description.

PR #2745 (2021-05-08): blif: Use library cells' start_offset and upto for wideports.
Fixes #2729.
commit 32a0ce9d (2021-06-18): blif: Use library cells' start_offset and upto for wideports.

Fixes #2729.
closed 2021-05-08 by mwkmwkmwk