YosysHQ/yosys #2739: write_cxxrtl: assertion error at -O5

jfng, 2021-04-28:
## Steps to reproduce the issue

```
module \top
  wire width 1 $foo
  wire width 1 \bar
  connect \bar $foo
end
```

```yosys repro.il -p "write_cxxrtl -O5"```

## Expected behavior

CXXRTL code is emitted.

## Actual behavior

```
ERROR: Assert `is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:905.
```

modwizcode, 2021-06-30:
Could this just be another variant of https://github.com/YosysHQ/yosys/issues/2540

whitequark, 2021-07-15:
@modwizcode This issue actually shows the reason why I felt uneasy about #2540. Apply it and you get another assert here.

jfng, 2021-07-16:
I still get an assertion error, with yosys checked out at 414154dd:
```
ERROR: Assert `wire_type.is_member()' failed in backends/cxxrtl/cxxrtl_backend.cc:2885.
```

whitequark, 2021-07-17:
This got broken again in 09218896d606726c7e3cd9caa635445694338fba.

whitequark, 2021-07-17:
So this is a kind of a pathological case: you have a public wire, which is an alias of a private wire, and both are optimized out completely. This is fine in `eval`, but since #2873, the public wire always appears in `debug_eval` (which is intentional and a good outcome), but there is no code to calculate its value.

whitequark, 2021-07-17:
@jfng Out of curiosity, how did you get to this minimized test case? I can't come up with an obvious way to get here from Verilog or nMigen source.

jfng, 2021-07-17:
I encountered this bug by trying to simulate a standalone LiteDRAM core (generated with https://github.com/enjoy-digital/litedram/blob/master/litedram/gen.py), and reduced it with bugpoint.

whitequark, 2021-07-17:
Can you check if it works with https://github.com/YosysHQ/yosys/pull/2879?

jfng, 2021-07-17:
#2879 has fixed it, thanks! The assertion error is gone and the memtest passes.

PR #2870 (2021-07-16): cxxrtl: mark dead local wires as unused even with inlining disabled
Fixes #2739.
commit 54b6cb64 (2021-07-16): cxxrtl: mark dead local wires as unused even with inlining disabled.

Fixes #2739.
PR #2879 (2021-07-17): cxxrtl: treat internal wires used only for debug as constants
Fixes #2739 (again).
commit 4aa65f40 (2021-07-17): cxxrtl: treat internal wires used only for debug as constants.

Fixes #2739 (again).
closed 2021-07-17 by whitequark