YosysHQ/yosys #2746: In CXXRTL edge eval is before calculating value

korneliuszo, 2021-05-06:
I have code: https://github.com/korneliuszo/nhci/tree/aa6cbe84b3df7d755972e31baf6ef20d714c9a23

and generated code doesn't trigger posedge in spi.v
	always @(posedge IOWR)
	begin
			IOWR_data = D_in;
	end

IOWR here needs calculation, when i delete it in 
https://github.com/korneliuszo/nhci/blob/aa6cbe84b3df7d755972e31baf6ef20d714c9a23/fw/top.v#L69
It latches value.

In generated code there is:
bool p_sim::eval() {
	bool converged = true;
	bool posedge_i_flatten_5c_top_2e__24_or_24_top_2e_v_3a_69_24_48__Y = this->posedge_i_flatten_5c_top_2e__24_or_24_top_2e_v_3a_69_24_48__Y();

and then
	i_flatten_5c_top_2e__24_or_24_top_2e_v_3a_69_24_48__Y = or_uu<1>(p_top_2e_IOWR, not_u<1>(p_top_2e_CONFIGURED));

so as i understand edge doesn't have a chance to be detected

korneliuszo, 2021-05-06:
Attaching unit test
[2746.tar.gz](https://github.com/YosysHQ/yosys/files/6433790/2746.tar.gz)
Should return 1 but returns 0.
After moving:
	bool posedge_i_and_24_top_2e_v_3a_13_24_1__Y = this->posedge_i_and_24_top_2e_v_3a_13_24_1__Y();
after
	p_C = i_and_24_top_2e_v_3a_13_24_1__Y;
it works

modwizcode, 2021-08-18:
So I've taken a look at this issue. First I'd like to cover a few issues in the design that are likely to bring trouble in general, and especially so combined with things like cxxrtl (and real hardware for that matter).

First of all, I would recommend never mixing blocking `=` assignments and non-blocking `<=` in a single block. In general I don't recommend blocking assignments at all (especially not in clocked blocks), they tend to infer latches easily and mostly make things harder. This is especially true when working with cxxrtl.

Secondly, doing `@(posedge signal)` when that signal is not a clock is also likely to produce bad behaviors. In this case I seem to notice you both look for a low-high transition using the `IOWR_ev` registers and (for `IOWR`) you seem to be trying to also use the posedge. 

I'm not sure why you've split them but that's not going to end well in general. In particular your block that uses the `IOWR` signal as a clock is not necessarily going to run with any relationship to the normal clock, which means that now you technically have two unsynchronized clock domains, which is why you never want to detect edges on non-clocks by using `@(posedge signal)`. The approach with the `_ev` variables is the right one here.

All that being said, I've tested this in the latest CXXRTL and either as a result of more recent changes, or chance. The design actually seems to work as intended, probably due to lucky coincidence. The output generated code is significantly different from the snippets you reference. 

As a result I think this bug is both not an issue with cxxrtl in this case and no longer even occurs and should likely be closed.

whitequark, 2021-08-18:
There are other similar/related CXXRTL issues, in particular https://github.com/YosysHQ/yosys/issues/2780, that need to be fixed by rewrting the scheduler. That would, if I recall correctly, also resolve this issu.

modwizcode, 2021-08-18:
@whitequark you probably know better than I, but it seems here that things are further complicated by the fact that there's actually two clock domains in this logic (likely unintentionally so). I don't think any amount of scheduling changes are going to account for the fact that one of those processes might very well run before the other. 

It's also equally possible this is a specific interaction of verilog semantics regarding the ordering of blocking assignments in various independent processes that I'm not remembering. 

That's a bit unrelated since I do agree we should rewrite the scheduler, but in this case it sounds like the issue was that things got optimized out that shouldn't have and that seems to no longer occur. If you want I can try to bisect this until I find a version that doesn't work?

modwizcode, 2021-08-18:
I should note that when I wrote my initial reply I wasn't able to download the unit test posted and assumed it was no longer present. Seems like that was not the case so I'm going to try testing again with the actual test files.

korneliuszo, 2021-08-19:
Code on master branch works as intended. (tested using https://github.com/korneliuszo/pcmciahost)

I need seperate clock domain as PCMCIA is asynchronous, and I think main clock is too slow to latch data after synchronization in main clock, so I use more clocks.

modwizcode, 2021-08-19:
So now that whatever issue with the download seems to have been resolved, been able to look at this testcase and confirm that yes, there's indeed a scheduling error occurring in the reduced testcase. So that does yes need to be addressed. 

Need to look into the precise reason this is happening but suspect it has something to do with the fact that the process block is using `@(posedge xyz)` on a signal that is computed within the logic. This issue should be solved by rewriting the scheduler to recognize the dependencies properly as wq said. Still going to look into this though to see why specifically this isn't noticed by existing code in the meantime.