YosysHQ/yosys #2751: Yosys gives wrong output - changing "assign a = b" to direct wire (in instantiation) works

jay20162016, 2021-05-11:
## Steps to reproduce the issue
Hello! My design at https://github.com/jay20162016/fomu_async is not working again.
Uncommenting the line with the "works" comment (and commenting the "doesn't work" lines) makes it work.
(Don't mix comments from different cases)
Similarly, the "doesn't work" lines don't work.
Here is the relevant part of the code:
```verilog
...
  swap #(.Rpol(Rpol), .N(N), .NATIVE(NATIVE)) main (
    r_i, a_i, d_i,
    r1_i, a1_i, d1_i,
    ctl_a, ctl_b, actl_i,
    r_o, a_o, d_o,
    // rt_o, rt_o, dt_o, // works
    rt_o, at_o, dt_o, // doesn't work case A+B
    rst
    );

  sink main_sink (
    rt_o, at_o
    ); // doesn't work case A
  // assign at_o = rt_o; // doesn't work case B
  // assign rt_o = at_o;
...
```
Sink is just a buffer module.

I have already verified that in yosys, the module-level output is the same, except for a few extra wires (the typical stuff).
However, the top-level output is different: https://gist.github.com/jay20162016/1aeaa09e75ac65907619ca580fa3ed8d

## Expected behavior

*Please describe the behavior you would have expected from the tool.*
Works, no significant difference in top-level output

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*
Doesn't work, see above.

See also https://github.com/YosysHQ/nextpnr/issues/699 (original issue there).

Thanks in advance!

clairexen, 2021-05-17:
Quoting gatecat from [the thread on ther nextpnr issue](https://github.com/YosysHQ/nextpnr/issues/699#issuecomment-839596926):

> I'm not sure if there is a bug in either tool. It would definitely be best if you could provide a more concrete example of what is going wrong.

jay20162016, 2021-05-17:
Whops, race condition on my side. :( async is full of these issues.

closed 2021-05-17 by jay20162016