YosysHQ/yosys #2760: operator *div* can't process cases where the width of output is greater then divisor

xjliang, 2021-05-18:
## Steps to reproduce the issue
input RTL:
[image]

## Expected behavior

y[2] = 0    y[1] = a[1] + !b   y[0] = a[0] + !b

## Actual behavior

y[2] = !b    y[1] = a[1] + !b   y[0] = a[0] + !b

xjliang, 2021-05-18:
we can fix this by update the [techmap.v](https://github.com/YosysHQ/yosys/blob/master/techlibs/common/techmap.v#L345)
to
[image]

, where we should set the width to the max of inputs, other than all ports.

mwkmwkmwk, 2021-05-18:
This is not a bug — the current behavior is correct.

The Verilog standard (IEEE 1364-2005 5.1.5) specifies that, if the second operand to a division or modulus operator is zero, the result value is just `'x`.  Since yosys is a synthesis tool, it is well within its rights to change that `'x` to an arbitrary value, such as all-ones that naturally follows from the division algorithm used by the techmap.

closed 2021-05-18 by mwkmwkmwk