YosysHQ/yosys #2761: deploy Yosys in nodejs

Bestduan, 2021-05-18:
Hello, these days I plan to deploy Yosys in VScode, but I have encountered some problems. The first is whether Yosys.js can be started and used in native JavaScript?
The following problems are encountered during debugging：
[image]
How can I export the functions in Yosys.js and use them in native JavaScript?
I look forward to your clarification.

racerxdl, 2021-06-11:
Since that's a web library, try just requiring it without assigning to anything.

```javascript
require("./yosysjs");
console.log(YosysJS);
```

In browser-js, things are declared under global.XXXX or window.XXXX. In case of nodejs it probably goes under global.XXXX which will maps directly as XXXXX.

You can also edit `yosys.js` file and add to the end:

```javascript
exports.YosysJS = YosysJS;
```

Which should make it work like you tried.

Bestduan, 2021-06-27:
First of all, Yosys.js does not contain YosysJS objects. In addition, after that, I tried to use the methods exposed in Yosys.js directly and successfully read the Verilog file and completed some designs. However, I encountered problems in how to export the design.I can't export the designed file.The specific code is as follows:
`
var fs = require("fs");
var yosys   = require("../yosys-no-wasm-0.9/yosys");
console.log(yosys);

// yosys.FS.writeFile("demo.v", fs.readFileSync("./demo.v", "utf-8"), { encoding: 'utf8' });
let demo_text = fs.readFileSync("D:/project/Code/Javascript/yosysjs/demo.v", "utf-8");
yosys.FS_createPath("/","verilog",true,true)
yosys.FS_createDataFile("/verilog","demo.v", demo_text,true,true,false);
yosys.ccall('run', '', ['string'], ["read_verilog /verilog/demo.v"]);
yosys.ccall('run', '', ['string'], ["synth_anlogic -top demo"]);
yosys.ccall('run', '', ['string'], ["write_verilog /verilog/full.v"]);
`
The first problems encountered are as follows：
[image]
How can I fix it?
Second, how do I get the contents of the Full.v file if I do it correctly?
Being able to run Yosys on NodeJS is really important for my future work. Looking forward to your reply.

closed 2021-07-02 by Bestduan