YosysHQ/yosys #2762: Double-equal (==) operator not working as intended

norandomtechie, 2021-05-19:
## Steps to reproduce the issue

I have two (intended-to-be) identical designs, with a key difference in syntax noted by a comment in each of the designs:

```
// broken.sv
`default_nettype none
// Empty top module

module top1 (
  input logic hz100, reset,
  output logic [7:0] right,
  output logic red
);

  // Your code goes here...
  always_ff @(posedge hz100, posedge reset)
  if (reset)
    right <= 0;
  else
    right <= clk ? 0 : right + 1;
  
  logic clk;
  assign clk = right == 8'd99;    // Different from fixed.sv
  always @(posedge clk, posedge reset)
  if (reset)
    red <= 0;
  else
    red <= ~red;

endmodule
```

```
// fixed.sv
`default_nettype none
// Empty top module

module top2 (
  input logic hz100, reset,
  output logic [7:0] right,
  output logic red
);

  // Your code goes here...
  always_ff @(posedge hz100, posedge reset)
  if (reset)
    right <= 0;
  else
    right <= clk ? 0 : right + 1;
  
  logic clk;
  assign clk = &(right ~^ 8'd99);   // Different from broken.sv
  always @(posedge clk, posedge reset)
  if (reset)
    red <= 0;
  else
    red <= ~red;

endmodule
```

Given these two designs, I synthesized each of them to produce s_fixed.sv and s_broken.sv with the following commands:

```
yosys -p "read_verilog -sv broken.sv; synth_ice40 -top top1; write_verilog s_broken.sv"
yosys -p "read_verilog -sv fixed.sv; synth_ice40 -top top2; write_verilog s_fixed.sv"
```

And simulate the resulting Verilog with IcarusVerilog for delay-supported gate-level simulations.  For reference, the testbench is given:

```
`timescale 1ns/1ns

module main;
    reg hz100;
    wire reset;
    reg [20:0] pb;
    wire [7:0] right1, right2;
    wire red1, red2;

    top1 ice401 (
        .hz100(hz100),
        .reset(reset),
        .right(right1),
        .red(red1)
    );
        
    top2 ice402 (
        .hz100(hz100),
        .reset(reset),
        .right(right2),
        .red(red2)
    );

    reset_on_start ros (
        .reset(reset), 
        .clk(hz100), 
        .manual(pb[3]&&pb[0]&&pb[16])
    );

    integer i = 0;

    initial begin
      $dumpfile("trace.vcd");
      $dumpvars(0, main);
      pb = 0;
      for (i = 0; i < 4; i++) begin
        hz100 = 1; #1; hz100 = 0; #1;   // clock until reset goes low
      end
      for (i = 0; i < 198; i++) begin
        hz100 = 1; #1; hz100 = 0; #1;   // clock 
        $display("i: %3d ---> right1: %3d, right2: %3d, red1: %1d, red2: %1d", i, right1, right2, red1, red2);
      end
    end
endmodule

module reset_on_start(
    output reset,
    input clk,
    input manual
);
  logic [2:0] startup = 0;
  assign reset = startup[2] | manual;    // MSB drives the reset signal
  always @ (posedge clk, posedge manual)
    if (manual == 1)
      startup <= 0;             // start with reset low to get a rising edge
    else if (startup == 0)
      startup <= 4;             // hold reset high for 4 cycles
    else if (startup == 4)
      startup <= 5;
    else if (startup == 5)
      startup <= 6;
    else if (startup == 6)
      startup <= 7;
    else if (startup == 7)
      startup <= 1;             // Final state is 1.
endmodule
```

## Expected behavior

What is intended is for the "red" output to toggle every time the value of "right" is 99, which is what occurs with a regular IcarusVerilog simulation:

```
iverilog -o sim.vvp tb_top.sv broken.sv fixed.sv -g2012
./sim.vvp
```

The behavior of both modules was entirely identical, with the output of the testbench looking like:

```
i:   0 ---> right1:   0, right2:   0, red1: 0, red2: 0
i:   1 ---> right1:   1, right2:   1, red1: 0, red2: 0
i:   2 ---> right1:   2, right2:   2, red1: 0, red2: 0
...
i:  98 ---> right1:  98, right2:  98, red1: 0, red2: 0
i:  99 ---> right1:  99, right2:  99, red1: 1, red2: 1
i: 100 ---> right1:   0, right2:   0, red1: 1, red2: 1
i: 101 ---> right1:   1, right2:   1, red1: 1, red2: 1
...
i: 195 ---> right1:  95, right2:  95, red1: 1, red2: 1
i: 196 ---> right1:  96, right2:  96, red1: 1, red2: 1
i: 197 ---> right1:  97, right2:  97, red1: 1, red2: 1
```

This behavior is what happens if I run Icarus directly on the source files and testbench, without using Yosys.

## Actual behavior

However, the behavior when using the synthesized Verilog, and the SB_* modules from the provided Yosys share/ice40 folder differs from that of the simulation with just the source code above.

For the gate-level simulation, there are two additional files to specify to Icarus - share/ice40/cells_sim.v and share/ice40/cells_map.v.  Since the current Yosys master versions of those files don't seem to work well with Icarus, I've attached my version of those files below (modified and retained from an older version of Yosys, although I can't recall which):

cells_sim.v: https://gist.github.com/norandomtechie/8dd95f09a246edf40791651fa8a7fd57
cells_map.v: https://gist.github.com/norandomtechie/069ec927449ca6f63f2fe1c3253c074e

The resulting command is:

```
iverilog -o sim.vvp tb_top.sv s_broken.sv s_fixed.sv cells_sim.v cells_map.v -g2012
./sim.vvp
```

Upon simulating, the following occurs instead:

```
i:   0 ---> right1:   0, right2:   0, red1: 0, red2: 0
i:   1 ---> right1:   1, right2:   1, red1: 0, red2: 0
i:   2 ---> right1:   2, right2:   2, red1: 0, red2: 0
...
i:  96 ---> right1:  96, right2:  96, red1: 0, red2: 0
i:  97 ---> right1:  97, right2:  97, red1: 0, red2: 0
i:  98 ---> right1:  98, right2:  98, red1: 1, red2: 0
i:  99 ---> right1:  99, right2:  99, red1: 0, red2: 1
i: 100 ---> right1:   0, right2:   0, red1: 0, red2: 1
i: 101 ---> right1:   1, right2:   1, red1: 0, red2: 1
i: 102 ---> right1:   2, right2:   2, red1: 0, red2: 1
i: 103 ---> right1:   3, right2:   3, red1: 0, red2: 1
...
i: 194 ---> right1:  94, right2:  94, red1: 0, red2: 1
i: 195 ---> right1:  95, right2:  95, red1: 0, red2: 1
i: 196 ---> right1:  96, right2:  96, red1: 0, red2: 1
i: 197 ---> right1:  97, right2:  97, red1: 0, red2: 1
```

Note above that red1 (which belongs to s_broken) goes high a cycle before red2 (which belongs to s_fixed), which does go high correctly when right1/right2 == 99.  Then, it somehow goes low again at 99, and remains that way, even though there is no visible reason for it to do so.

For some inexplicable reason, the s_broken.sv file does not yield the same design as s_fixed.sv, with the only difference in their original source code being that the broken one uses `right == 8'd99` while the fixed one uses `&(right ~^ 8'd99)`.  It appears as though the use of the == sign is resulting in different behavior.

The version of Yosys used the latest Git master at the time of writing (also recompiled Yosys with "-Os" to ensure it wasn't an optimization issue): 
`Yosys 0.9+4052 (git sha1 d8c5d681, gcc 9.3.0-17ubuntu1~20.04 -fPIC -O3)`

gatecat, 2021-05-19:
I'm not sure if this is the cause of the issues in simulation, but using unregistered combinational comparisons as a clock is bound to fail in hardware, as neither  FPGA synthesis nor PnR guarantee a glitch free implementation.

norandomtechie, 2021-05-19:
Registering `clk` worked.  I completely blanked on that.

What about the fact that I can do `&(right ~^ 8'd49)` and still get it to work?  That makes me wonder if there really is an issue with Yosys here.  Would it have to do with how the '==' sign is synthesized?

gatecat, 2021-05-19:
> What about the fact that I can do assign clk = &(right ~^ 8'd49) without registering it and still get it to work? That makes me wonder if there really is an issue with Yosys here. Would it have to do with how the '==' sign is synthesized?

Random good luck, I suspect

udif, 2021-05-19:
> What about the fact that I can do `assign clk = &(right ~^ 8'd49)` without registering it and still get it to work? 

Just looking at this line makes me cringe.

Even if it seems to you this works, you will find months later that it doesn't work when your board gets too hot or too cold and your error would be too subtle to detect at first if you will ever get to the root cause at all.

If you want your designs to just work, never assign your clock signal with an expression. It should originate from a PLL block or an external pin, and propagate your design through module port hierarchy connections only.
More advanced designs may use multiple clock domains where it is necessary to sync to multiple external clocks (think USB<>Ethernet), or use clock gating techniques to save power. When you need to do these tricks, your tools must support this, you must clearly declare this, work with specific methodologies (adding synchronizers and clock gating blocks) to make it work, and run specific tests to make sure you haven't had any mistake, and still this is one of the trickiest areas in digital design.

Never mess with your single clock unless you absolutely have to. There are enough pitfalls in digital design as it is.

mwkmwkmwk, 2021-05-19:
Closing bug, as this kind of asynchronous logic is just bound to behave weird by laws of digital circuit design

closed 2021-05-19 by mwkmwkmwk