YosysHQ/yosys #2806: Getting Syntax Error while running make command

meetm81193, 2021-05-30:
After doing make config-clang and make config-gcc as per the README.md file. While running make command getting syntax error as below

frontends/blif/blifparse.cc: In function ‘void Yosys::parse_blif(Yosys::RTLIL::Design*, std::istream&, std::string, bool, bool, bool)’:
frontends/blif/blifparse.cc:296:25: note: synthesized method ‘Yosys::RTLIL::Const& Yosys::RTLIL::Const::operator=(const Yosys::RTLIL::Const&)’ first required here 
[ 11%] Building frontends/ilang/ilang_parser.tab.cc
frontends/ilang/ilang_parser.y:48.20-44: syntax error, unexpected {...}
make: *** [frontends/ilang/ilang_parser.tab.cc] Error 1

And the installation is getting failed.

I am using CENTOS 6.10 version of OS with bash SHELL. Have installed all the prerequistes like gwak, python3 etc.

zachjs, 2021-06-01:
The referenced file `frontends/ilang/ilang_parser.y` was renamed back in August. Do you run into issues trying to build a more recent version too? Also, what version of `bison` are you using?

meetm81193, 2021-06-02:
I am trying to install yosys-0.9 and currently installed bison version is 2.4.1

zachjs, 2021-06-02:
Bison v2.4.1 was released in 2008. I don't know what our exact version requirements are, but upgrading to a more recent version should help.

zachjs, 2021-07-13:
Given the inactivity, I am closing this issue. Please feel free to reopen if you have an update!

closed 2021-07-13 by zachjs