YosysHQ/yosys #2810: Build fails with HEAD (13b901bf1c5) on macos

jaroslov, 2021-05-31:
1. Clone the repo;
2. cd into the repo;
3. `make -j`;
4. Fails with the following message:

```
[ 55%] Building passes/sat/sim.o
frontends/rtlil/rtlil_parser.y:48.9-18: syntax error, unexpected identifier, expecting string
make: *** [frontends/rtlil/rtlil_parser.tab.cc] Error 1
make: *** Waiting for unfinished jobs....
```

mwkmwkmwk, 2021-05-31:
Seen that one before, you need to get the compilation process to use a modern version of bison.

closed 2021-05-31 by mwkmwkmwk

jaroslov, 2021-05-31:
Yep. Just figured out how crazy ancient Bison is on macOS.

jaroslov, 2021-05-31:
In case this gets anyone else, the fix for a macOS build environment is trivial:

nakengelhardt, 2021-05-31:
Yosys also comes with a Brewfile - if you have brew, the recommended way to set up the environment for compiling is:
```
$ brew tap Homebrew/bundle && brew bundle
```