YosysHQ/yosys #2811: transparent nMigen Memory triggers yosys assertion

anuejn, 2021-05-31:
## Steps to reproduce the issue

```python
from nmigen import *
from nmigen.back import verilog
mem = Memory(width=8, depth=8)
m = Module()
m.submodules += mem.read_port(domain="comb", transparent=True)
m.submodules += mem.write_port()
verilog.convert(m)
```

rtlil:
```
attribute \generator "nMigen"
attribute \top 1
attribute \nmigen.hierarchy "top"
module \top
  wire width 3 input 0 \mem_r_addr
  wire width 1 input 1 \clk
  wire width 1 input 2 \mem_w_en
  wire width 3 input 3 \mem_w_addr
  wire width 8 input 4 \mem_w_data
  memory width 8 size 8 \mem
  cell $meminit $1
    parameter \MEMID "\\mem"
    parameter \ABITS 4
    parameter \WIDTH 8
    parameter \WORDS 8
    parameter \PRIORITY 0
    connect \ADDR 4'0000
    connect \DATA 64'0000000000000000000000000000000000000000000000000000000000000000
  end
  wire width 8 \mem_r_data
  cell $memrd \U$$0
    parameter \MEMID "\\mem"
    parameter \ABITS 3
    parameter \WIDTH 8
    parameter \CLK_ENABLE 0
    parameter \CLK_POLARITY 1
    parameter \TRANSPARENT 1
    connect \CLK 1'0
    connect \EN 1'1
    connect \ADDR \mem_r_addr
    connect \DATA \mem_r_data
  end
  cell $memwr \U$$1
    parameter \MEMID "\\mem"
    parameter \ABITS 3
    parameter \WIDTH 8
    parameter \CLK_ENABLE 1
    parameter \CLK_POLARITY 1
    parameter \PRIORITY 0
    connect \CLK \clk
    connect \EN { { \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en } }
    connect \ADDR \mem_w_addr
    connect \DATA \mem_w_data
  end
end
```

## Expected behavior

The nmigen code is successfully generated to verilog.

## Actual behavior

A yosys assertion is triggered:
```
nmigen._toolchain.yosys.YosysError: ERROR: Assert `!port.transparent' failed in kernel/mem.cc:293.
```

(see https://github.com/nmigen/nmigen/issues/617)

anuejn, 2021-05-31:
This is a regression; it worked previously, and now it is broken. 
Probably related to https://github.com/YosysHQ/yosys/commit/c7076495f197732725456992c9a02aed9966139a ?

mwkmwkmwk, 2021-05-31:
*sigh*

mwkmwkmwk, 2021-05-31:
Thanks for the report, will be fixed soon.

mwkmwkmwk, 2021-05-31:
By the way, there is no such thing as a "transparent" asynchronous read port, and it never behaved differently from a plain one.  It just so happens that a check function meant to trigger on invalid memory configs was added, and caught this.

PR #2812 (2021-06-01): kernel/mem: Recognize some deprecated memory port configs.
Transparency is meaningless for asynchronous ports, so we assume
transparent == false to simplify the code in this case.  Likewise,
enable is meaningless, and we assume it is const-1.  However,
turns out that nMigen emits the former, and Verilog frontend emits
the latter, so squash these issues when ingesting a $memrd cell.

Fixes #2811.
commit 6d5d8457 (2021-06-01): kernel/mem: Recognize some deprecated memory port configs.

Transparency is meaningless for asynchronous ports, so we assume
transparent == false to simplify the code in this case.  Likewise,
enable is meaningless, and we assume it is const-1.  However,
turns out that nMigen emits the former, and Verilog frontend emits
the latter, so squash these issues when ingesting a $memrd cell.

Fixes #2811.
closed 2021-06-01 by mwkmwkmwk