YosysHQ/yosys #2815: How to extract differemt nets, i.e. Scalar Net & Bus Net seperately connected to cells

xjliang, 2021-06-06:
Is there a way to traverse the connections of the cell and then extract the name of nets connected to `ports?
```cpp
for (auto conn_it : cell->connections()) {
    // conn_it.first -> port name
    // conn_it.second -> SigSpec,
}
```
I want to know how to handle the **SigSpec** of the ports? In other words, how to extract the net name from it?
I just cannot understand the difference between **SigChunk** and **SigBit** of **SigSpec**. And when should we convert the ** SigSpec** to **SigChunk** or **SigBit**, **Wire**?
Can I do like this:
```cpp
// Given a SigSpec corresponding to a cell connection, returns the wire name connected to the cell port
std::string extract_netname(RTLIL::SigSpec sig) {
  RTLIL::Wire* wire = chunk->wire;

  if (sig.is_chunk()) {
    RTLIL::SigChunk chunk = sig.as_chunk();
    wire = chunk->wire;
  } else if (sig.is_bit()) {
    RTLIL::SigBit bit = sig.as_bit();
    wire = bit.wire;
  } else {
    // Here is what ? Bus Net? how can we extract the net name?
  }

  std::string result;
  if (wire) {
      result = wire->name;
  }
  return result;
}
```
As the code shown above, how can I extract the wire if sig is neither SigChunk nor SigBit? When this happens, should I return a list of wires?

thcoura, 2021-07-05:
I comprehend your goal, but I would like to understand what you want to achieve with the names. Would be a graph with the net names?

xjliang, 2021-07-14:
I just want to convert RTLIL to another data structure.

closed 2021-08-13 by mmicko