YosysHQ/yosys #2826: `extract` pass treats outputs of a sub-circuit as inputs (may be a bug)

nindanaoto, 2021-06-12:
As mentioned in #2370,  `extract_fa` does not extract half adder cells, so I tried to extract half adder cells by defining a half adder cell in a Verilog file and calling `extract` pass in the script.
This method detected half adder cells correctly, but the outputs of half adders are treated as inputs of half adder sub-circuits.
The figure shows a part of the schematic which includes extracted half adder (ha) cells.
The attached ZIP file includes the necessary files to reproduce the result. Run `docker build -t extractbug && docker run --rm -it --net host -e DISPLAY=$DISPLAY -v $HOME/.Xauthority:/root/.Xauthority extractbag bash` and run `/yosys/yosys build.ys` in the shell.

[image]
[extractbug.zip](https://github.com/YosysHQ/yosys/files/6642025/extractbug.zip)

nindanaoto, 2021-06-12:
Contents of the zip file in plaintexts:
Mul.v
```
module Mul(
  input        clock,
  input        reset,
  input  [3:0] io_a,
  input  [3:0] io_b,
  output [7:0] io_out
);
  assign io_out = io_a * io_b; // @[mul.scala 12:17]
endmodule
```
ha.v
```
module ha (
  input a, b,
  output sum,carry
);
  assign sum = a^b;
  assign carry = b&&a;
endmodule
```
build.ys
```
# read subcircuit
read_verilog ha.v

# elaborate design hierarchy
hierarchy -check -top ha

# the high-level stuff
# proc; opt; fsm; opt; memory; opt

# mapping to internal cell library
techmap; opt

# mapping logic to gates.
abc -g gates,MUX,NMUX

# save and reset
design -stash ha
design -reset-vlog

# read design 
read_verilog Mul.v

# elaborate design hierarchy
hierarchy -check -top Mul

# the high-level stuff
proc; opt; fsm; opt; memory; opt

# mapping to internal cell library
techmap; opt

# mapping logic to gates.
abc -g gates,MUX,NMUX

# extract full adder
extract_fa -fa

# extract half adder
extract -map %ha -verbose

# cleanup
clean -purge

# print statistics
stat

#check output circuit
show
```
Dockerfile
```'FROM ubuntu:20.04

RUN apt-get update && apt-get -y upgrade

RUN DEBIAN_FRONTEND=noninteractive apt-get install -y build-essential clang bison flex \
	libreadline-dev gawk tcl-dev libffi-dev git \
	graphviz xdot pkg-config python3 libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev

RUN git clone https://github.com/YosysHQ/yosys.git

WORKDIR yosys

RUN make

WORKDIR /mul

COPY build.ys /mul/
COPY Mul.v /mul/
COPY Mul.v /mul/
COPY ha.v /mul/
```

povik, 2025-03-19:
> The output of show reflects this. It's only for builtin cells that the directions are shown in an intuitive way.

That's not accurate -- the direction is shown if a module definition is available.

> This is because the design of RTLIL is heavily inspired by Verilog which doesn't really require any sort of "header file" or "forward declaration" of instantiating a module

That's not true either. Some tools tolerate unknown modules, usually as a configurable option.

widlarizer, 2025-03-19:
Ah right, thanks for the clarification