YosysHQ/yosys #2828: How to modify Verilog defines?

rodrigomelo9, 2021-06-13:
Hi. I tried several ways to change Verilog defines (based on the Yosys doc and a few examples that I found through the internet), without success. Could this feature be broken? I am doing something wrong?

## Steps to reproduce the issue

[verilog_defines_issue.zip](https://github.com/YosysHQ/yosys/files/6644825/verilog_defines_issue.zip)

I provided a simple description and a bash script with all the things that I tried (using an updated Docker container).
Of course, I dealt with this problem in another description, where I saw no changes in the synthesized result.

## Expected behavior

```
-- Running command... --

1. Executing Verilog-2005 frontend: defines.v
Parsing Verilog input from `defines.v' to AST representation.
Generating RTLIL representation for module `\Defines'.
aaa = 2
bbb = 3
Successfully finished Verilog frontend.
```

## Actual behavior

```
-- Running command... --

1. Executing Verilog-2005 frontend: defines.v
Parsing Verilog input from `defines.v' to AST representation.
Generating RTLIL representation for module `\Defines'.
aaa = 1
bbb = 2
Successfully finished Verilog frontend.
```

Which are the default values, not my modifications.

Regards,
Rodrigo

mwkmwkmwk, 2021-06-13:
It's supposed to work that way — the defines start out as (2, 3), but are redefined to (1, 2) while the frontend is working through the file.

If you want the outside defines to have precedence, just have this at the top of your file instead:

```
`ifndef aaa
`define aaa 1
`endif
`ifndef bbb
`define bbb 2
`endif
```

closed 2021-06-13 by mwkmwkmwk

rodrigomelo9, 2021-06-14:
Got it. Thanks!