YosysHQ/yosys #2833: Initial block with $readmemh innefective with Lattice ECP5-25F FPGA

tambewilliam, 2021-06-15:
With Lattice ECP5-25F FPGA used by OrangeCrab, $readmemh does not actually initialize an array element.

Also, attempting to manually initialize an array element beyond its number of elements, passes synthesis instead of failing.

## Steps to reproduce the issue

parameter SZ = 256;
parameter DW = 128;
parameter SRCFILE = "mem.hex";

reg [DW-1:0] u [0:SZ-1];

initial begin
	if (SRCFILE != "") begin
		$readmemh (SRCFILE, u);
	end
end

gatecat, 2021-06-15:
You're going to need to give quite a bit more information here (in particular, the design and contents of mem.hex) in order for anyone to be able to help.

tambewilliam, 2021-06-15:
Certainly; I will provide fast response; find attached mem.hex that was used; I had to append '.txt' to the filename for github to accept it as attachment.
[mem.hex.txt](https://github.com/YosysHQ/yosys/files/6657073/mem.hex.txt)

jotego, 2021-06-15:
I use something very similar for synthesis in Cyclone FPGAs (for example, [here](https://github.com/jotego/jtframe/blob/b90e3876f7d8f43283bbf363a7b81fd433a75965/hdl/ram/jtframe_prom.v#L105)). Although the `$readmemh` may look like something for simulation, it is certainly vital for synthesis too.

tambewilliam, 2021-06-15:
@jotego I agree that it is vital.

gatecat, 2021-06-15:
It should be supported, but you still haven't provided enough information for someone to investigate what the actual problem is.

tambewilliam, 2021-06-15:
It would help to know which additional information to provide.

To speak more on the issue, $readmemh is simply not initializing the memory it receives as argument.
I tried to debug further replacing $readmemh with manually initializing memory as such:
```
initial begin
	u[0]    <= 64'00add0391034d0ad;
	u[1024] <= 64'6e69206d6172206e; // Indexing element beyond size of array which is 256.
end
```

Not only was the initialization not taking place, synthesis would succeed despite indexing an element of the array beyond its size.

Please let me know any additional information you would like me to provide.

nakengelhardt, 2021-06-15:
Please provide a complete design that can be passed to `synth_ecp5` to produce the faulty netlist so that we can take a look at what is happening. There is no module in your example at all, and without the surrounding context the memory would just be optimized away.

tambewilliam, 2021-10-10:
Apologies for the long delay in responding. I finally found time to write code to reproduce the issue.

Find attached in [orangecrab02.tar.gz](https://github.com/YosysHQ/yosys/files/7317591/orangecrab02.tar.gz) following files:
```
orangecrab02
 |-orangecrab02.pcf
 |-orangecrab02.v
 |-yosys
 | |-makefile
 | |-mem.hex
 ```
 
 Build using:
 ```
 cd orangecrab02/yosys/
 make
 ```
 
 Program the orangecrab using:
 ```
 make prog
 ```
 
 The logic in `orangecrab02.v` initializes an array with bit sequences that are played on an LEDs every 1/4 seconds.
 A different bit sequence in the array gets selected when pressing the reset button.
 
 I find that the array gets initialized differently when manually initializing it versus using $readmemh: snippet from attached code:
 ```
 initial begin
	//u[0] <= 128'haaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;
	//u[1] <= 128'h0a0a0a0a0a0a0a0a0a0a0a0a0a0a0a0a;
	$readmemh (SRCFILE, u);
end
 ```
 
 Initialization of the array does happen, but I see that with $readmemh, the blinking of the LED does not always reflect the initialization value.

tambewilliam, 2021-10-10:
I tried the latest version of the tools from https://github.com/YosysHQ/oss-cad-suite-build which is oss-cad-suite-windows-x64-20211010.

I see a much better behavior; the older version of the tool I was using definitely had issues.

To see the issue even better, I have modified the logic to AND corresponding bits between a manually initialized array u1 and an array u2 initialized using `readmemh`, as shown below:

```
module orangecrab02 (

	 usr_btn_n

	,clk48mhz_i

	,led_red_n
	,led_green_n
	,led_blue_n
);

input wire clk48mhz_i;

input wire usr_btn_n;

output wire led_red_n;
output wire led_green_n = 1'b1;
output wire led_blue_n = 1'b1;

localparam CLKFREQ = 48000000/4 /* 48 MHz */;

localparam SZ = 2;
localparam DW = 128;
parameter SRCFILE = "mem.hex";

reg [DW-1:0] u1 [0:SZ-1];
initial begin
	u1[0] <= 128'haaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;
	u1[1] <= 128'h0a0a0a0a0a0a0a0a0a0a0a0a0a0a0a0a;
end

reg [DW-1:0] u2 [0:SZ-1];
initial begin
	$readmemh (SRCFILE, u2);
end

reg [1-1:0]  idx;
reg [32-1:0] cntr;
reg [DW-1:0] r1;
reg [DW-1:0] r2;

assign led_red_n = (~r1[0] && ~r2[0]);

reg saved_usr_btn_n; // Used for edge detection.

always @ (posedge clk48mhz_i) begin
	saved_usr_btn_n <= usr_btn_n;
	if (!usr_btn_n && usr_btn_n != saved_usr_btn_n) begin
		r1 <= u1[idx];
		r2 <= u2[idx];
		idx <= idx + 1'b1;
	end else if (cntr >= CLKFREQ) begin
		r1 <= {r1[0], r1[DW-1:1]};
		r2 <= {r2[0], r2[DW-1:1]};
		cntr <= 0;
	end else begin
		cntr <= cntr + 1'b1;
	end
end

endmodule
```

With the latest version of the tools, u1 and u2 are correctly identical, while with the older version of the tool I was using, u1 and u2 are not identical as they should.

I am continuing to run tests, as I am still seeing issues with a larger design that uses `readmemh`; it could be another issue with the design itself.