YosysHQ/yosys #2841: CXXRTL: nmigen AsyncFIFO triggers assert when generated from verilog (rtlil works).

hansfbaier, 2021-06-18:
## Steps to reproduce the issue
Extract this file:
[gray-cxxrtl-bug.tar.gz](https://github.com/YosysHQ/yosys/files/6675954/gray-cxxrtl-bug.tar.gz)

change into the directory and execute:
`yosys bug.ys`

## Expected behavior
No assert is triggered and the output file is generated.

## Actual behavior

An assert will be triggered and an empty output file is generated:
```
[...]
Module `asyncfifodemo' contains feedback arcs through wires:

Module `gray' contains feedback arcs through wires:
  gray.$13
  gray.$7
  gray.$5
  gray.$3
  gray.$1
  gray.$11
  $flatten\gray.$xor$gray.v:47$3_Y
Warning: Design contains feedback wires, which require delta cycles during evaluation.
$flatten\gray.$xor$gray.v:47$3_Y <=================
[gray-cxxrtl-bug.tar.gz](https://github.com/YosysHQ/yosys/files/6675957/gray-cxxrtl-bug.tar.gz)
 This is the name of the wire that triggers the assert (debug print added)
ERROR: Assert `is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:906.
```
Note: When generating from RTLIL like this:
```
read_rtlil gray.il
write_cxxrtl gray.cc
```
then everything works fine.

modwizcode, 2021-06-30:
I suspect like a few other bugs this might be the same root cause, can you try applying https://github.com/YosysHQ/yosys/pull/2541 and let me know if this still happens. If not I have another theory that I will try and get around to testing.

hansfbaier, 2021-07-02:
@modwizcode I get merge conflicts, when I try to cherry-pick your commits onto master. Your fork is 330 commits behind yosys master. I will try to checkout your fork then.

hansfbaier, 2021-07-02:
@modwizcode I checked out your fork, and yes, that fixes the issue.

hansfbaier, 2021-07-02:
This is a duplicate of https://github.com/YosysHQ/yosys/issues/2540 and tested fixed by https://github.com/YosysHQ/yosys/pull/2541

modwizcode, 2021-07-02:
> @modwizcode I get merge conflicts, when I try to cherry-pick your commits onto master. Your fork is 330 commits behind yosys master. I will try to checkout your fork then. 

I'll have to manually pull them up to date. 

I'm glad it worked though, hopefully those changes are indeed the fix.

whitequark, 2021-07-15:
Let's keep this open for now, at least the repro is a lot smaller than in #2540.

whitequark, 2021-07-15:
Minimized with bugpoint:
```
module \gray$1
  wire $xor$repro1.v:47$3_Y
  wire \$9
  wire width 8 \o
  cell $xor $xor$repro1.v:47$3
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \o [3]
    connect \B 1'x
    connect \Y $xor$repro1.v:47$3_Y
  end
  connect \$9 $xor$repro1.v:47$3_Y
  connect \o [2] \$9
end
```

PR #2871 (2021-07-16): cxxrtl: don't mark buffered internal wires as UNUSED for debug
Public wires may alias buffered internal wires, so keep `BUFFERED` wires in debug information even if they are private. Debug items are only created for public wires, so this does not otherwise affect how debug information is emitted.

Fixes #2540.
Fixes #2841.
Supersedes #2541.
commit 44a3d924 (2021-07-16): cxxrtl: don't mark buffered internal wires as UNUSED for debug.

Public wires may alias buffered internal wires, so keep BUFFERED
wires in debug information even if they are private. Debug items are
only created for public wires, so this does not otherwise affect how
debug information is emitted.

Fixes #2540.
Fixes #2841.
closed 2021-07-16 by whitequark