YosysHQ/yosys #2843: Error parsing SystemVerilog struct accessor within case statements

TheOneKevin, 2021-06-22:
Failure with "failed to detect width for identifier" when parsing struct accessors within switch statements.

## Steps to reproduce the issue

Smallest breaking example:
```systemverilog
typedef struct packed {
    logic a;
} mystruct;

module top();
    mystruct s;
    always @(*) case(s.a)
        default: begin end
    endcase
endmodule
```

Version output:
```bash
$ yosys --version
Yosys 0.9+4081 (git sha1 862e84eb, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

## Expected behavior

No compilation error. Support for packed struct is assumed from the README:

> [...] packed structs and unions are supported.

## Actual behavior

Error produced incorrectly
```
1. Executing Verilog-2005 frontend: hs32_decode.sv
Parsing SystemVerilog input from `hs32_decode.sv' to AST representation.
Generating RTLIL representation for module `\top'.
hs32_decode.sv:7: ERROR: Failed to detect width for identifier \s.a!
```

zachjs, 2021-06-22:
I have a good idea what's going wrong here. I'll work to push a fix tomorrow.

TheOneKevin, 2021-06-22:
Thank you, this was fast!

PR #2845 (2021-07-15): sv: fix two struct access bugs
- preserve signedness of struct members
- fix initial width detection of struct members (e.g., in case expressions)

Fixes #2843
closed 2021-07-15 by zachjs