YosysHQ/yosys #2852: Bring back the version update bot with GitHub Actions

thoughtpolice, 2021-07-05:
As of earlier this year, there was a nice robot called "yosys-bot" which, every 24 hours, would run `make bumpversion` on the repository and commit the result, if and only if a commit was made in that time. That is no longer the case, and the bot has been defunct for several months (and, for the sake of both brevity and keeping discussions on topic, I won't further speculate or reason as to why.)

However, I would like this bot to return in some form. I am the maintainer of Yosys for Nixpkgs, and that robot's autotagging was very useful. I would regularly update the latest version of the yosys package with version numbers based on those version bump commits. They provided a very useful "line in the sand" which I could use to push and scope updates around. In fact, our build system specifically checks that the version number of the Nix package and the version in `YOSYS_VER` matches as a sanity check to ensure they're correct. As of now, nixpkgs is using a build from February, ` 0.9+4052`, which was the last version bump that was performed if I'm correct.[1]

So it would be nice if we could bring it back, and since everything these days is consolidating around GitHub Actions, I think it's quite viable to just have a bot that runs every 24 hours and does this. Actually, 24 hours might be too frequent; one week would also be fine.

Of course, the thing is, this kind of work is very boring and annoying and has a few tiny sharp edges and because you can't test GH Actions you normally do this thing where you push to a "testing" branch 400,000 times in order to get all the little bits right. So it's easy, but also annoying at the same time. Alas. But the basic idea is just:

1. Run at some interval
2. Run `make versionbump`
3. Commit the result under some identity

Again, this is useful for me and I haven't seen anyone else complain. So perhaps I'll get around to doing this myself when I feel like it...

---

[1] There is of course another point here which is "why don't you just use the stable release", but in general I find yosys to be extremely stable in general and the features since the 0.9 release are significant; I'm a user of yosys myself and this means that reverting back to a stable release is a serious regression in terms of features and support. And as a user and a package maintainer, this creates a kind of positive feedback loop; as a result I think the yosys package in Nixpkgs is very high quality and easy to use (along with several others I maintain, like `nextpnr`). Finally, users of `nixpkgs` are able to use those packages on any system, so many people use the latest tip of the `master` branch to get very fresh packages, `yosys` included.

(There are also projects like YoWASP; but there seems to already be infrastructure in place for doing nightly builds there so the only thing that would change here is having a more accurate version number.)

So I think regardless of the situation with a new stable release, the autotagging bot is still quite useful.

whitequark, 2021-07-05:
That would be great to have!

whitequark, 2021-07-05:
> As of now, nixpkgs is using a build from February, ` 0.9+4052`, which was the last version bump that was performed if I'm correct.[1]

Looks like that last happened manually 2 months ago: https://github.com/YosysHQ/yosys/commit/25de8faf10157ab0cb40f77c7cbf3143527c598e

zachjs, 2021-07-05:
Is the git repo itself available when these releases are generated? The makefile already runs git (with a fallback), so I wonder how far we could get by computing the version "always" without having it hardcoded, or by using `git describe` (I also find `--dirty` useful during development).

I have sv2v (via Haskell shenanigans) building in the `git describe` version, and using the "release version" as a fallback if the git repo isn't available at bios time.

thoughtpolice, 2021-07-05:
No, Nix does not allow the git repository to be present when a package is built &mdash; in general there are arguments about whether or not it's acceptable to make the VCS history a requirement for your build. But aside from that, allowing it causes other problems; it would ruin reproducibility for a number of reasons (e.g. you immediately have to begin thinking full vs shallow clones, whether or not to fetch all references,, and combined with the fact you can prune/compact history it just makes getting rid of it and avoiding the whole problem way easier.)

Of course you might argue it's all a bit roundabout, because I have to literally go into the nixpkgs repository and open the `yosys` package description in my editor in order to commit updates. At that time I could, for instance, run `(cd $HOME/src/yosys && git log --format=oneline | wc)` and just write the version number into the right place myself. But the matching of the package version with `YOSYS_VER` is really an anti-stumbling block to make sure things are done correctly by other maintainers as well, who might not be as familiar with the package as me. It also helps ensure that yosys *users* can trace a package back to the right version in the repository by using that version number alone. And finally, that result is also good for yosys *developers* &mdash; because if you start getting bug reports for versions that don't actually align with what is in the repository history, it can make debugging and analysis significantly more confusing.

zachjs, 2021-07-05:
I figured the motivations were along those lines. I appreciate you making it clear for me!

mmicko, 2021-07-27:
Fixed with https://github.com/YosysHQ/yosys/commit/45968ad740d3d060592adb37d8a378e602c7522b

closed 2021-07-27 by mmicko

thoughtpolice, 2021-07-27:
Thanks a ton! A [new update for Nixpkgs is cooking now](https://github.com/NixOS/nixpkgs/pull/131720) to celebrate...