YosysHQ/yosys #2865: FDCE equivalence issue

jaromharris, 2021-07-13:
I'm using Yosys for checking equivalence between verilog netlists, and I've noticed that any netlists containing FDCE xilinx primitives fail to prove equivalence. I've provided a zip file
[fdce_test.zip](https://github.com/YosysHQ/yosys/files/6810383/fdce_test.zip)
 with a gold.v, gate.v, and commands.ys containing the Yosys commands I have been using. The expected result is a message saying equivalence is successfully proven, but instead the following error is thrown:

```
9. Executing EQUIV_STATUS pass.
Found 1 $equiv cells in equiv:
  Of those cells 0 are proven and 1 are unproven.
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$2005: \fdce_gold_gold.Q \fdce_gate_gate.Q
Found a total of 1 unproven $equiv cells.
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

Is this an issue with the commands I am using, or is it a bug in the Yosys code?

mwkmwkmwk, 2021-07-13:
You need to use either `clk2fflogic` or `async2sync` pass before equivalence checking whenever asynchronous logic is involved.

jaromharris, 2021-07-13:
Okay, thank you!

closed 2021-07-13 by jaromharris

jaromharris, 2021-07-14:
@mwkmwkmwk Is asynchronous logic simply not supported in Yosys equivalence checking?