YosysHQ/yosys #2867: Unexpected macro behavior

abakst, 2021-07-15:
## Steps to reproduce the issue

`Pad.sv` contains two almost identical modules, the difference is that the vector indices use a `MIN` macro in one case, and the macro is manually expanded in the other case. Loading the Verilog produces warnings in the `MIN` case.
[testcase.zip](https://github.com/YosysHQ/yosys/files/6819463/testcase.zip)

The problem is reproducible with:

`$ yosys -q -p "read_verilog -sv Pad.sv"`

## Expected behavior

No warnings.

## Actual behavior

Displays:

```
Pad.sv:6: Warning: Range [63:0] select out of bounds on signal `\y': Setting 63 MSB bits to undef.
Pad.sv:6: Warning: Ignoring assignment to constant bits:
    old assignment: { 63'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx \y } = { 63'000000000000000000000000000000000000000000000000000000000000000 \x }
    new assignment: \y = \x.
```

mwkmwkmwk, 2021-07-15:
I have investigated this and... it turns out that the inner MIN macro invocation's arguments somehow overwrite outer invocation's arguments.

Yes, apparently our macro expansion implementation is non-reentrant.

zachjs, 2021-07-15:
Fortunately, the issue isn't non-reentrancy in the traditional sense, but that the preprocessor appears to have no concept of when a particular macro invocation begins or ends. I have an idea for a not-too-painful fix if you haven't already started on this, @mwkmwkmwk.

mwkmwkmwk, 2021-07-15:
Go for it, then.

abakst, 2021-07-21:
> Fortunately, the issue isn't non-reentrancy in the traditional sense, 

Glad to hear that 😄

PR #2868 (2021-07-29): verilog: save and restore overwritten macro arguments
Fixes #2867
closed 2021-07-29 by zachjs