YosysHQ/yosys #2877: cxxrtl - debug_item could use get<> and set<>

galibert, 2021-07-16:
debug_item C++ interface is currently minimal.  Support for get<> and set<> as values have would be rather nice and would at the same time avoid people messing up between curr and next.

Perhaps it should have the same interface as value<>, but value has a big interface which may not be that useful for external code piloting the simulation.

whitequark, 2021-07-18:
The interface will look like this:
```cxx
// toggle clock
bool clk = debug["clk"].get<1, bool>();
debug["clk"].set<1>(!clk);
```

You will have to provide the expected width of the value/wire referenced by the debug item, and it is checked (using an assert, in debug builds) against the actual width stored in the debug item.

whitequark, 2021-07-18:
See #2880.

PR #2880 (2021-07-18): cxxrtl: add debug_item::{get,set}
Fixes #2877.
commit 948fc10d (2021-07-18): cxxrtl: add debug_item::{get,set}.

Fixes #2877.
closed 2021-07-18 by whitequark