YosysHQ/yosys #2886: Equivalence checking issue

jaromharris, 2021-07-21:
I am learning to use Yosys equivalence checking to compare verilog netlists. I've created a minimal design that fails equivalence checking when compared against itself. In this zip folder [not_equivalent.zip](https://github.com/YosysHQ/yosys/files/6858598/not_equivalent.zip) I have a .ys script that compares not_equivalent.v with itself. This yields the following output:
```
16. Executing EQUIV_STATUS pass.
Found 1 $equiv cells in equiv:
  Of those cells 0 are proven and 1 are unproven.
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$6061: \lut6_inst.s5_2[3]_gold \lut6_inst.s5_2[3]_gate
Found a total of 1 unproven $equiv cells.
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```
The original verilog design (original_design.v) is also included in the zip folder. 

Also, here is a zip folder [equivalent.zip](https://github.com/YosysHQ/yosys/files/6858619/equivalent.zip) that is a similar design that passes as equivalent.

Is there a Yosys pass that I'm missing in my .ys file?

jaromharris, 2021-07-29:
@mwkmwkmwk When you have a moment, could you provide me with some assistance here?

mwkmwkmwk, 2021-07-29:
Don't run `prep` after `equiv_make`

closed 2021-07-29 by mwkmwkmwk

jaromharris, 2021-08-04:
@mwkmwkmwk Thank you again for your help! Here, [not_equivalent.zip](https://github.com/YosysHQ/yosys/files/6934097/not_equivalent.zip) I have the the same netlist as before named `not_equivalent_gold.v`, and I am comparing it to `not_equivalent_gate.v`. The only difference between the 2 files is the names of the internal wires and the names of the primitives. I have modified the script to not run `prep` after `equiv_make`. Please let me know when you have taken a look at it, or if you could point me to some resources that would help me as I run into more problems with equivalence checking it would be much appreciated!