YosysHQ/yosys #2887: cxxrtl blackbox behavior issue / possible bug ?

smunaut, 2021-07-24:
Attached is an example showing the issue.  (sorry about tar.bz2 inside zip ... github doesn't want tar.bz2 attach :/ )

[cxxrtl_bb.zip](https://github.com/YosysHQ/yosys/files/6872767/cxxrtl_bb.zip)

It's a simple memory block implemented two ways : once using pure verilog  ( `mem_sim_data_verilog.v` ) and the other as a `cxxrtl` blacbox ( `blackbox.cpp` & `mem_sim_data_bb.v` ) and I expect them both to produce the same behavior, but they don't.

I didn't even get to test the write port but the read port appears as a "async" read instead of a "sync" one (i.e. data appears directly and not at the next cycle like I'd expect).

whitequark, 2021-07-24:
This is a bug; there is currently no way to make the blackbox version work.

whitequark, 2025-08-03:
No, I left the job that would've paid for it. I will do it when porting cxxrtl to prjunnamed.