YosysHQ/yosys #2888: Long runtime (endless) for yosys to build for EPC5 LFE5U-25F BG256

Waxpple, 2021-07-25:
## Steps to reproduce the issue

I am using iCESugar-Pro (LFE5U-25F BG256).
Here is my code.
After type `python3 blinky.py`, It stuck at command yosys -q forever with 100% cpu usage in htop.
```
import os
import subprocess
from typing import List
from nmigen import Elaboratable, Module, Signal
from nmigen.build import *
from nmigen.build.run import LocalBuildProducts
from nmigen.cli import main_parser, main_runner
from nmigen.vendor.lattice_ecp5 import *

class Blinker(Elaboratable):
    def __init__(self):
        pass
    def elaborate(self, platform: Platform) -> Module:
        m = Module()

        clk_freq = platform.default_clk_frequency

        timer = Signal(int(clk_freq // 2), reset=int(clk_freq // 2) - 1)
        led = platform.request("led").o

        with m.If(timer == 0):
            m.d.sync += timer.eq(timer.reset)
            m.d.sync += led.eq(~led)
        with m.Else():
            m.d.sync += timer.eq(timer -1)
        
        return m
    
    def ports(self) -> List[Signal]:
        return []

class Board(LatticeECP5Platform):
    device = "LFE5U-25F"
    package = "BG256"
    speed = "6"
    default_clk = "clk1"
    default_rst = "rst"
    resources = [
        Resource("clk1", 0, Pins("P6",dir="i"), Clock(25e6),
            Attrs(GLOBAL=True, IO_STANDARD = "LVCMOS33")),
        Resource("led", 0, Pins("B11",dir="o"),
            Attrs(GLOBAL=True, IO_STANDARD = "LVCMOS33")),
        Resource("rst", 0, Pins("L14",dir="i"),
            Attrs(GLOBAL=True, IO_STANDARD = "LVCMOS33")),
    ]
    connectors = []
    def toolchain_program(self, products, name):  
        iceprog = os.environ.get("ICEPROG", "iceprog")
        with products.extract("{}.bin".format(name)) as bitstream_filename:
            subprocess.check_call([iceprog, bitstream_filename])
if __name__ == "__main__":
    Board().build(Blinker(), do_program= False)
```
I do install the nextpnr-epc5 and project trellis.

## Expected behavior
I am following the tutorial on [https://www.youtube.com/watch?v=AQOXoKQhG3I&list=LL&index=3&t=3898s](https://www.youtube.com/watch?v=AQOXoKQhG3I&list=LL&index=3&t=3898s)
build time should be very short for blinky.
## Actual behavior
Never complete the build.

Do I do something wrong here? 
Thank you

Xiretza, 2021-07-25:
With latest nmigen ([abb2642](https://github.com/nmigen/nmigen/commit/abb264225644a8030e6221c49a74c8c07611d132)), I get the following error:

```
nmigen.back.rtlil.ImplementationLimit: Wire created at /tmp/blinky.py:18 is 12500000 bits wide, which is unlikely to synthesize correctly
```

Waxpple, 2021-07-25:
oh .. Maybe the youtube tutorial is on older version nmigen.. after changing the code I get only warnings!
```python
import os
import subprocess
from typing import List
from nmigen import Elaboratable, Module, Signal
from nmigen.build import *
from nmigen.build.run import LocalBuildProducts
from nmigen.cli import main_parser, main_runner
from nmigen.vendor.lattice_ecp5 import *

class Blinker(Elaboratable):
    def __init__(self):
        pass
    def elaborate(self, platform: Platform) -> Module:
        led   = platform.request("led", 0)
        timer = Signal(20)

        m = Module()
        m.d.sync += timer.eq(timer + 1)
        m.d.comb += led.o.eq(timer[-1])
        return m
    
    def ports(self) -> List[Signal]:
        return []

class Board(LatticeECP5Platform):
    device = "LFE5U-25F"
    package = "BG256"
    speed = "6"
    default_clk = "clk1"
    default_rst = "rst"
    resources = [
        Resource("clk1", 0, Pins("P6",dir="i"), Clock(25e6),
            Attrs(GLOBAL=True, IO_STANDARD = "LVCMOS33")),
        Resource("rst", 0, Pins("L14",dir="i"),
            Attrs(GLOBAL=True, IO_STANDARD = "LVCMOS33")),
        Resource("led", 0, Pins("B11",dir="o"),
            Attrs(IO_STANDARD = "LVCMOS33")),
    ]
    connectors = []
    def toolchain_program(self, products, name):  
        iceprog = os.environ.get("ICEPROG", "iceprog")
        with products.extract("{}.bin".format(name)) as bitstream_filename:
            subprocess.check_call([iceprog, bitstream_filename])
if __name__ == "__main__":
    Board().build(Blinker(), do_program= False)
```
I will try to solve the problem!
```python
Warning: IOBUF 'led_0__io' attribute 'IO_STANDARD' is not recognised (on line 5)
Warning: IOBUF 'clk1_0__io' attribute 'GLOBAL' is not recognised (on line 7)
Warning: IOBUF 'clk1_0__io' attribute 'IO_STANDARD' is not recognised (on line 7)
Warning: IOBUF 'rst_0__io' attribute 'GLOBAL' is not recognised (on line 9)
Warning: IOBUF 'rst_0__io' attribute 'IO_STANDARD' is not recognised (on line 9)
5 warnings, 0 errors
```
Thank you.

closed 2021-07-25 by Waxpple

gatecat, 2021-07-25:
`IO_STANDARD` should be `IO_TYPE` on ECP5, and `GLOBAL` removed altogether.

Waxpple, 2021-07-25:
> `IO_STANDARD` should be `IO_TYPE` on ECP5, and `GLOBAL` removed altogether.

Everything is `GLOBAL=True` by default in ECP5?

gatecat, 2021-07-25:
There's no distinction between IO buffer types for global/non global signals, and global buffer insertion is done automatically by PnR.

Waxpple, 2021-07-25:
> There's no distinction between IO buffer types for global/non global signals, and global buffer insertion is done automatically by PnR.

Oh! got it. Thank you
I am new to this, it is very nice to have your help!
After I fire the `python3 blinky.py`, does it actually do yosys+next-pnr+ecppack and all I have to do left is program the bitstream onto FPGA?