YosysHQ/yosys #2890: Is the way to control the $ in net namess

dineshannayya, 2021-07-26:
I see  Yosys tool does Two dimensional register array changed with $ in net name like:  u_wb2sdrc.u_cmdfifo.mem**$**rdreg[0]**$**d[0]

Is there is way to do renaming Rule inside yosys to avoid $ in the netname?

OpenSTA tool is having issue in handling net with $.
[sdrc_top.synthesis_optimized.v.gz](https://github.com/YosysHQ/yosys/files/6876019/sdrc_top.synthesis_optimized.v.gz)

mwkmwkmwk, 2021-07-28:
The naming rules cannot be changed, but you can change the names after the fact to `_<number>_` by running `rename -enumerate`, avoiding the problem

whitequark, 2021-08-09:
@mwkmwkmwk This is a public name; so `rename -enumerate` won't do anything to it.
[image]

dineshannayya, 2021-08-09:
Is there a plan to fix this net naming with $ issue ..  As this $ will be
propagated inside the .spef file and Open STA tool is not handing this.

When i raised the issue with Open STA tool, they say's "is illegal to have
$ in net name"

mwkmwkmwk, 2021-08-09:
May I ask which version of yosys are you using? I explicitely fixed the particular piece of code that emitted `$rdreg` to make a private wire name a long time ago.

dineshannayya, 2021-08-09:
I was using "Yosys 0.9+4081 (git sha1 b6721aa, gcc 8.3.1 -fPIC -Os)"
inside the openlane docker ..

I see there is a new version of yosys available, i will cross-check on the
same.

Yosys 0.9+4249 (git sha1 d8fcf1ab2, clang 10.0.0-4ubuntu1 -fPIC -Os)

dineshannayya, 2021-08-10:
This issue not observed in the latest yosys 0.9+4249 .. I am closing this ticket

closed 2021-08-10 by dineshannayya