YosysHQ/yosys #2891: Verilog backend: pack identical signals in concatenation

flaviens, 2021-07-27:
Hi!

In some cases, some source code or internal Yosys passes generate long arrays of signal copies.
It would be great of Yosys could pack them back when writing Verilog.

For instance for the trivial source code:

```verilog
module mymodule # (
  parameter int unsigned Width = 16
) (
	input logic a_i,
	output logic [Width-1:0] y_o
);
  assign y_o = {Width{a_i}};
endmodule
```

Applying the Yosys commands:
```
yosys> read_verilog -sv mymodule.sv
yosys> hierarchy -top mymodule
yosys> write_verilog -noattr yosys_out.v
```

Yields:
```verilog
module mymodule(a_i, y_o);
  input a_i;
  output [15:0] y_o;
  assign y_o = { a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i, a_i };
endmodule
```

It would be extremely beneficial to get back instead something such as:
```verilog
module mymodule(a_i, y_o);
  input a_i;
  output [15:0] y_o;
  assign y_o = {16{a_i}};
endmodule
```

The main motivation is that Verilator will complain if there are too many tokens in the same line.

Thanks for your time! :slightly_smiling_face:

whitequark, 2024-01-19:
The code to do this is present in the CXXRTL backend, and could potentially be shared.

povik, 2024-01-22:
> The code to do this is present in the CXXRTL backend, and could potentially be shared.

FWIW the `show` comand happens to have code for this too