YosysHQ/yosys #2894: YosysHQ Manual Down

BracketMaster, 2021-07-27:
When I check July 24, 2021, the yosys manual at Yosys HQ was two years
old. Now its simply offline: https://yosyshq.net/yosys/files/yosys_manual.pdf

I keep one [here](https://yehowshuaimmanuel.com/yosys_manual.pdf) that
I've built for personal reference, but would be good to get the YosysHQ one 
back up.

gatecat, 2021-07-27:
It would be nice, if possible, if we could get manuals built automatically by the same infrastructure as the nightly builds and pushed somewhere?

mmicko, 2021-07-28:
This is now handled by github action on https://github.com/YosysHQ/yosys-manual-build and produces manual.pdf
@mwkmwkmwk can you please update link to on site to point to https://github.com/YosysHQ/yosys-manual-build/releases/download/manual/manual.pdf (link will not change) ? there is also presentation.pdf if needed

mwkmwkmwk, 2021-07-28:
Link updated

mmicko, 2021-07-28:
Great. Thanks

closed 2021-07-28 by mmicko