YosysHQ/yosys #2896: `memory` pass bug on simple RAM

flaviens, 2021-07-29:
Hello!

First I'd like to address you a big thank you. Yosys is a very powerful tool and bugs seem quite rare, great job!

I tried to apply the `memory` pass of Yosys on a simple RAM, and observed that after `memory; proc; opt -purge`, it was completely optimized out.

Here is the SystemVerilog code for the RAM (simplified from the [opentitan prim_generic_sram_1p](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim_generic/rtl/prim_generic_ram_1p.sv)):

```verilog
module data_mem (
	clk_i,
	write_i,
	addr_i,
	wdata_i,
	wmask_i,
	rdata_o
);
	parameter unsigned [8:0] Width = 1;
	parameter unsigned [8:0] Depth = 4;
	localparam unsigned [8:0] Aw = $clog2(Depth);
	input wire clk_i;
	input wire write_i;
	input wire [Aw - 1:0] addr_i;
	input wire [Width - 1:0] wdata_i;
	input wire [Width - 1:0] wmask_i;
	output reg [Width - 1:0] rdata_o;
	reg [Width - 1:0] mem [Depth];
	always_ff @(posedge clk_i) begin
			if (write_i) begin
				reg unsigned [8:0] i;
				for (i = 0; i < Width; i = i + 1)
					if (wmask_i[i])
						mem[addr_i][i] <= wdata_i[i];
			end
			else
				rdata_o <= mem[addr_i];
    end
endmodule
```

`stat` gives:

```
=== data_mem ===

   Number of wires:                 24
   Number of wire bits:             53
   Number of public wires:           8
   Number of public wire bits:      17
   Number of memories:               1
   Number of memory bits:            1
   Number of processes:              1
   Number of cells:                  1
     $memrd                          1
```

The `memory` pass followed by a Verilog backend gives the code below. You can observe that `rdata_o` is never influenced by the input. **This is not expected**.

```
module data_mem(clk_i, write_i, addr_i, wdata_i, wmask_i, rdata_o);
  reg \initial  = 0;
  reg [1:0] _00_;
  reg _01_;
  reg _02_;
  reg _03_;
  reg [8:0] _04_;
  reg [1:0] _05_;
  reg _06_;
  reg _07_;
  reg [8:0] _08_;
  reg [1:0] _09_;
  reg _10_;
  reg _11_;
  wire _12_;
  reg [1:0] _13_;
  reg _14_;
  reg _15_;
  input [1:0] addr_i;
  input clk_i;
  output rdata_o;
  reg rdata_o;
  reg [8:0] \sv2v_autoblock_1.i ;
  input wdata_i;
  input wmask_i;
  input write_i;
  always @* begin
    if (\initial ) begin end
    _03_ = rdata_o;
    _04_ = _08_;
    _00_ = _05_;
    _01_ = _06_;
    _02_ = _07_;
    casez (write_i)
      1'h1:
        begin
          _05_ = _09_;
          _06_ = _10_;
          _07_ = _11_;
          _08_ = 9'h001;
          casez (wmask_i)
            1'h1:
              begin
                _09_ = addr_i;
                _10_ = wdata_i;
                _11_ = 1'h1;
              end
            default:
              begin
                _09_ = 2'hx;
                _10_ = 1'hx;
                _11_ = 1'h0;
              end
          endcase
        end
      default:
        begin
          _08_ = \sv2v_autoblock_1.i ;
          _05_ = 2'hx;
          _06_ = 1'hx;
          _07_ = 1'h0;
          _03_ = _12_;
        end
    endcase
  end
  always @(posedge clk_i) begin
      rdata_o <= _03_;
      \sv2v_autoblock_1.i  <= _04_;
      _13_ <= _00_;
      _14_ <= _01_;
      _15_ <= _02_;
  end
  assign _12_ = 1'hx;
endmodule
```

Eventually, continuing with `proc; opt -purge` unsurprisingly optimizes the logic out because of the issue with `memory`:
```verilog
module data_mem(clk_i, rst_ni, write_i, addr_i, wdata_i, wmask_i, rdata_o);
  input [1:0] addr_i;
  input clk_i;
  output rdata_o;
  input wdata_i;
  input wmask_i;
  input write_i;
  assign rdata_o = 1'hx;
endmodule
```

Changing `reg [Width - 1:0] mem [Depth]` into `reg [Depth-1:0][Width - 1:0] mem` is a successful workaround, which also suggests that the issue comes from the `memory` pass.

Thanks! :slightly_smiling_face:

gatecat, 2021-07-29:
Hi! 

The `memory` passes work on elaborated netlists, rather than RTLIL processes, so you need to run `proc` before `memory`.

Running `proc; memory; opt -purge` does appear to work fine (it no longer gets entirely optimised away), although I haven't looked at the synthesis result in detail.

flaviens, 2021-07-29:
Excellent, thank you for your very fast answer, this works like a charm! :slightly_smiling_face:

closed 2021-07-29 by flaviens