YosysHQ/yosys #2907: ECP5 (ULX3S): std::out_of_range assertion failure

tcal-x, 2021-08-03:
## Steps to reproduce the issue

Origin: LiteX build in litex_boards/targets/: `./radiona_ulx3s.py --build --load`

To reproduce: unzip the zipfile, enter the `gateware/` directory, run `bash build_radiona_ulx3s.sh`.

[yosys_range_error.zip](https://github.com/YosysHQ/yosys/files/6920710/yosys_range_error.zip)

## Expected behavior

yosys finishes without error, and the build script continues with nextpnr-ecp5

## Actual behavior

yosys exits with an error and the following messages:
```
  Checking rule #6 for bram type $__ECP5_DP16KD (variant 5):
    Bram geometry: abits=14 dbits=1 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_DP16KD: awaste=10445 dwaste=0 bwaste=10445 waste=10445 efficiency=36
    Rule for bram type $__ECP5_DP16KD (variant 5) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Selecting best of 6 rules:
    Efficiency for rule 4.5: efficiency=36, cells=32, acells=1
    Efficiency for rule 4.4: efficiency=72, cells=16, acells=1
    Efficiency for rule 4.3: efficiency=72, cells=16, acells=2
    Efficiency for rule 4.2: efficiency=85, cells=12, acells=3
    Efficiency for rule 4.1: efficiency=85, cells=12, acells=6
    Efficiency for rule 1.1: efficiency=85, cells=12, acells=12
    Selected rule 4.2 with efficiency 85.
    Mapping to bram type $__ECP5_DP16KD (variant 2):
      Read port #0 is in clock domain \VexRiscv.IBusCachedPlugin_cache.clk.
        Mapped to bram port B1.1.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 32) >= this->size() (which is 32)
build_radiona_ulx3s.sh: line 3: 1868932 Aborted                 (core dumped) yosys -l radiona_ulx3s.rpt radiona_ulx3s.ys
```

tcal-x, 2021-08-03:
Yosys ID:
`Yosys 0.9+4241 (git sha1 ca8ad6269, clang 10.0.0-4ubuntu1 -fPIC -Os)`

gatecat, 2021-08-03:
Another testcase if it helps, run this through `synth_nexus`:

```verilog
module ram0(
    // Write port
    input wrclk,
    input [7:0] di,
    input wren,
    input [5:0] wraddr,
    // Read port
    input rdclk,
    input rden,
    input [5:0] rdaddr,
    output reg [7:0] do);

    (* syn_ramstyle = "block_ram" *) reg [7:0] ram[0:63];

    initial begin
        ram[0] = 8'b00000001;
        ram[1] = 8'b10101010;
        ram[2] = 8'b01010101;
        ram[3] = 8'b11111111;
        ram[4] = 8'b11110000;
        ram[5] = 8'b00001111;
        ram[6] = 8'b11001100;
        ram[7] = 8'b00110011;
        ram[8] = 8'b00000010;
        ram[9] = 8'b00000100;
    end

    always @ (posedge wrclk) begin
        if(wren == 1) begin
            ram[wraddr] <= di;
        end
    end

    always @ (posedge rdclk) begin
        if(rden == 1) begin
            do <= ram[rdaddr];
        end
    end

endmodule
```

mwkmwkmwk, 2021-08-03:
Right, my bad

PR #2909 (2021-08-03): memory_bram: Move init data swizzling before other swizzling.
Fixes #2907.
commit 8733e192 (2021-08-03): memory_bram: Move init data swizzling before other swizzling.

Fixes #2907.
closed 2021-08-03 by mwkmwkmwk