YosysHQ/yosys #2908: Verilog frontend: array of packed structures in packed structure

flaviens, 2021-08-03:
Hi!

It seems that Yosys cannot parse this package correctly:

```verilog
package my_package;

  typedef struct packed {
    logic some_field;
  } first_struct_t;

  typedef struct packed {
    first_struct_t [1:0] some_other_field; // Yosys throws an error here
  } second_struct_t;

endpackage
```

Yosys throws the following error (while sv2v and Verilator succeed):
```
ERROR: syntax error, unexpected '[', expecting TOK_ID
```

[I created a repository for you to try.](https://github.com/flaviens/yosys_packed_struct_issue).

Thanks!

daglem, 2023-08-09:
This is a duplicate of #2677.

flaviens, 2023-08-09:
Nice catch! I'll close this one then :+1:

closed 2023-08-09 by flaviens